openhwfoundation/cvw #655: Failing derived floating-point configurations

davidharrishmc, 2024-03-07:
The following derived floating-point configurations are not yet passing regression-nightly:

vsim -c -do "do wally-batch.do fh_rv64gc arch64zfh"
  hangs on     "rv64i_m/Zfh/src/fcvt.h.l_b25-01.S",

f_rv32gc_arch32f
fh_rv32gc_arch32f
f_rv64gc_arch64f
f_rv64gc_arch64f
all fail on fmul_b6 sim = 0, signature = 3
(Kevin Kim looking for root cause)

rv64gc_arch64zfad: fcvtmod.w.d not yet working (Jordan Carlin troubleshooting)
   fround/froundx not yet implemented (Jessica Liu and Jordan Carlin writing)

jordancarlin, 2024-03-14:
fcvtmod.w.d fixed with #674. Issue still remains with Sail for b22 test.

davidharrishmc, 2024-03-15:
vsim -c -do "do wally-batch.do fh_rv64gc arch64zfh" hangs because arch64zfh fcvtmod.h.l test program performs a fsd, which is not supported in fh.  Consider modifying riscv-arch-test so that fcvtmod instructions that don't depend on d don't use it.

jordancarlin, 2024-03-25:
fcvtmod.w.d Sail issue with b22 tests resolved in riscv/sail-riscv/pull/442.

davidharrishmc, 2024-04-03:
Notes from Jordan about fmul failures:
https://docs.google.com/document/d/1p7zb4Vvd1LMBLRgEpXjHyp7etCaFaiBVrBZJM8jediE/edit

davidharrishmc, 2024-04-03:
Invoke with:
vsim -do "do wally.do f_rv32gc arch32f"

davidharrishmc, 2024-04-04:
Found a solution to the fmul failures.

jordancarlin, 2024-04-29:
https://github.com/riscv/sail-riscv/pull/442 was merged and Sail should now produce the correct signature.

davidharrishmc, 2024-04-29:
> riscv/sail-riscv#442 <https://github.com/riscv/sail-riscv/pull/442> was merged and Sail should now produce the correct signature.

Whoo!

davidharrishmc, 2024-05-13:
Closed with PR #794.

closed 2024-05-13 by davidharrishmc