openhwfoundation/cvw #662: No Definition for CacheHit in loggers.sv

Karl-Han, 2024-03-11:
The code refactoring in `src/cache/cache.sv` causes problem in the `testbench/common/loggers.sv` as it is still using the old `CacheHit` instead of `Hit`.
> Comparison: https://github.com/openhwgroup/cvw/compare/d1a1345e4d88b1afa16754ecd1554d71aadfe796...main?diff=split&w=#diff-60372075881c9299a8f886a0b6845b408bc1af35097e8ac7a071dd914a6a718eL85

```
time verilator -GTEST="\"arch64i\"" --prof-cfuncs -O3 -DVERILATOR=1 \                                                                               
--timescale "1ns/1ns" --timing --binary --top-module testbench "-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../
testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes                                                                             
%Error: ../testbench/common/loggers.sv:156:54: Can't find definition of 'CacheHit' in dotted variable: 'dut.core.ifu.bus.icache.icache.CacheHit'    
  156 |       HitMissString = dut.core.ifu.bus.icache.icache.CacheHit ? "H" :                                                                       
      |                                                      ^~~~~~~~                                                                               
%Error: ../testbench/common/loggers.sv:181:54: Can't find definition of 'CacheHit' in dotted variable: 'dut.core.lsu.bus.dcache.dcache.CacheHit'    
  181 |       HitMissString = dut.core.lsu.bus.dcache.dcache.CacheHit ? "H" :                                                                       
      |                                                      ^~~~~~~~                                                                               
%Error: Exiting due to 2 error(s)                  
```

davidharrishmc, 2024-03-25:
Can you just change CacheHit to Hit in loggers.sv?

davidharrishmc, 2024-03-28:
Closed with PR698.

closed 2024-03-28 by davidharrishmc