openhwfoundation/cvw #696: verilate errors

solomatnikov, 2024-03-28:
At d03ab77a11bd14fdd3a4e53dcc9559cea0f238de

```
./verilate 
Verilating rv64gc arch64zba
%Error: ../testbench/common/loggers.sv:156:54: Can't find definition of 'CacheHit' in dotted variable: 'dut.core.ifu.bus.icache.icache.CacheHit'
  156 |       HitMissString = dut.core.ifu.bus.icache.icache.CacheHit ? "H" :
      |                                                      ^~~~~~~~
%Error: ../testbench/common/loggers.sv:181:54: Can't find definition of 'CacheHit' in dotted variable: 'dut.core.lsu.bus.dcache.dcache.CacheHit'
  181 |       HitMissString = dut.core.lsu.bus.dcache.dcache.CacheHit ? "H" :
      |                                                      ^~~~~~~~
%Error: Exiting due to 2 error(s)
Exiting after rv64gc arch64zba verilation due to errors or warnings
```

solomatnikov, 2024-03-28:
duplicate of #662

davidharrishmc, 2024-03-28:
Closed with PR698

closed 2024-03-28 by davidharrishmc