openhwfoundation/cvw #697: Wally core hangs on a BEQ

adnan-hamid, 2024-03-28:
This issue was discovered while bringing up the Breker RISCV System VIP tests on the Wally design. 
Core appears to hang early in a basic checkout test, before actual testing started.

Waveform view:
[image]

Pretty doc with details:
[wally_issue_description.pdf](https://github.com/openhwgroup/cvw/files/14792500/wally_issue_description.pdf)

C source, Makefile and objdumps
[wally_issue.zip](https://github.com/openhwgroup/cvw/files/14792519/wally_issue.zip)

ross144, 2024-03-28:
Hi Adnan,

The zip archive is missing two header files.  Can you attach them? I'll be happy to debug.

stineje, 2024-03-28:
Hi Adnan,

I found the makefile.inc but I believe the trek_target is missing:

make: Warning: File 'test_mem.c' has modification time 28708 s in the future
riscv64-unknown-elf-gcc -march=rv64imfdczicbom -mabi=lp64d -Wa,-alhs -Wa,-L -mcmodel=medany  -mstrict-align -O2 -I../crt0 -c -o OBJ/test_mem.o test_mem.c > OBJ/test_mem.list
test_mem.c:15:10: fatal error: /home/nambi/trek5-2.0.10-GCC6_el7/target/include/trek_target_common.h: No such file or directory
   15 | #include "/home/nambi/trek5-2.0.10-GCC6_el7/target/include/trek_target_common.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [makefile.inc:56: OBJ/test_mem.o] Error 1

Let us know.  Take care.

All my best,

James

junambi, 2024-03-29:
Elf files for the test, attached.

Nambi
[elf_files.zip](https://github.com/openhwgroup/cvw/files/14800037/elf_files.zip)

adnan-hamid, 2024-04-01:
Hi James and Rose, 
Just making sure you connected that the two .elf files that Nambi posted belong to the test we posted. 
One passes and one fails.
-adnan

ross144, 2024-04-01:
I believe I've fixed the issue.  Pull request #703 allowed -O2 to run.

junambi, 2024-04-02:
I had fixed this locally to enable us to run with External memory and we saw the fail after that also.

I will rerun the simulation to see if that fixed the issue.

Nambi

junambi, 2024-04-02:
With this fix the test is passing with -O2 and -Og flags.

Seeing another issue with another test. Analyzing.

Please close this issue, will open another one as required.

Thank you,
Nambi

ross144, 2024-04-02:
Our convention is the original author of the issue closes the issue. Adnan do you mind closing?
I am happy to take a look at the other issue too.

closed 2024-04-03 by adnan-hamid