openhwfoundation/cvw #701: Comparison of Testbench between master branch version and posedge(signal) version

Karl-Han, 2024-03-29:
@davidharrishmc , this is the verilator performance comparison between the one on mater branch and the posedge(signal) version.

Command:

```shell
# compile non-profiling version:
time verilator -GTEST="\"arch64i\"" -O3 -DVERILATOR=1 --no-trace-top --stats \
--timescale "1ns/1ns" --timing --binary --top-module testbench \
"-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes

# run executable
time ./obj_dir/Vtestbench
```

## Runtime for master branch version
Here is the patch to avoid ERROR when calling `stop` at the end of testbench (I am not making PR because it may cause problem as it is also included in another PR): 

```diff
diff --git a/testbench/testbench.sv b/testbench/testbench.sv
index ee725c24..fe18debb 100644
--- a/testbench/testbench.sv
+++ b/testbench/testbench.sv
@@ -374,7 +374,11 @@ module testbench;
       if (test == tests.size()) begin
         if (totalerrors == 0) $display("SUCCESS! All tests ran without failures.");
         else $display("FAIL: %d test programs had errors", totalerrors);
+`ifdef VERILATOR // this macro is defined when verilator is used
+        $finish;
+`else
         $stop; // if this is changed to $finish, wally-batch.do does not go to the next step to run coverage
+`endif
       end
     end
   end
```

Result: 
```
TEST is arch64i
rv64i_m/I/src/add-01.S succeeded.  Brilliant!!!
...
rv64i_m/I/src/xori-01.S succeeded.  Brilliant!!!
SUCCESS! All tests ran without failures.
- ../testbench/testbench.sv:378: Verilog $finish
439.20user 0.10system 7:19.31elapsed 99%CPU (0avgtext+0avgdata 326656maxresident)k
0inputs+0outputs (0major+80746minor)pagefaults 0swaps
```
## Runtime for posedge(signal) version

In order to run this version, you can

- either resolve the conflict by `git cherry-pick bce63406dc94db14dbde0344139d175af8f5d184`
- or get `[pick_eliminate](https://github.com/Karl-Han/cvw/tree/pick_eliminate)`

Result:

```
TEST is arch64i
rv64i_m/I/src/add-01.S succeeded.  Brilliant!!!
...
rv64i_m/I/src/xori-01.S succeeded.  Brilliant!!!
SUCCESS! All tests ran without failures.
- ../testbench/testbench.sv:371: Verilog $finish
3.57user 0.12system 0:03.69elapsed 99%CPU (0avgtext+0avgdata 327168maxresident)k
0inputs+0outputs (0major+80740minor)pagefaults 0swaps
```

davidharrishmc, 2024-04-03:
Notes on this change:
Code changes integrated and still working in Questa, runtime unchanged for Questa

time verilator -GTEST="\"arch64i\"" --timescale "1ns/1ns" --timing --binary --top-module testbench "-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes
real    0m40.567s
user    0m35.152s
sys     0m5.290s

time obj_dir/Vtestbench 
rv64i_m/I/src/xori-01.S succeeded.  Brilliant!!!
SUCCESS! All tests ran without failures.
- ../testbench/testbench.sv:384: Verilog $finish

real    0m3.900s
user    0m3.696s
sys     0m0.196s

davidharrishmc, 2024-04-03:
Leaving SelectTest in the posedge clk block changes runtime to 3.935s
Restoring Validate to posedge clk block makes tests extremely slow

davidharrishmc, 2024-04-04:
Closed by PR704

closed 2024-04-04 by davidharrishmc