openhwfoundation/cvw #707: Testfloat on Verilator not progressing

davidharrishmc, 2024-04-04:
I started writing sim-testfloat-verilator.  It now compiles but seems to hang in the first test.

harris@chips:~/cvw/sim$ ./sim-testfloat-verilator 
harris@chips:~/cvw/sim$ time obj_dir/Vtestbenchfp 
Running ui32_to_f128_rne.tv vectors 
...(no progress)

davidharrishmc, 2024-05-15:
Now invokes with  
wsim -s verilator -t testbench_fp fdqh_ieee_rv64gc cvtint

Still hangs at
Running ui32_to_f128_rne.tv vectors

davidharrishmc, 2024-06-27:
After cleaning up some lint issues, 

wsim --tb testbench_fp --sim verilator rv32gc add

fails with

unning f64_add_rne.tv vectors 

Error in f64_add_rne.tv
TestNum          0 VectorNum          0 OpCtrl 6
inputs: ffffffff3f908000 0000000000000000 0000000000000000
SrcA: 00000000
 Res: ffffffff00000000 00
 Expected: 000000000007ffff 01

The reason is that readvectors is triggered (@VectorNum) before Unit, Fmt, OpCtrl) have their correct values, so the wrong vector is read.  In contrast, Questa receives

# TestNum          0 VectorNum          0 OpCtrl 6
# inputs: b68ffff8000000ff 3ff0000000000000 3f9080000007ffff
# SrcA: xxxxxxxx
#  Res: 3f9080000007ffff 01
#  Expected: 3f9080000007ffff 01

davidharrishmc, 2024-07-15:
Made some progress with commit 459eaaef6a5819b8a98063117582701684b43ca9
However, the sequencing of testbench_fp is highly messed up.  The module will likely require an entire refactoring to fix this.  Readvectors may need to be removed as a module and incorporated into the main module.  In the same always_ff @(posedge clk) block that loads vectors or advances the vector number, the readvectors logic may need to assign the new inputs and expected output.

davidharrishmc, 2024-10-10:
Defer this to an enhancement.  Modify wsim to produce errors on testbench_fp when run in a sim other than Questa.

davidharrishmc, 2026-09-18:
no longer part of a normal flow.

jordancarlin, 2026-11-07:
Might be worth revisiting this if we think the testfloat testbench still has value. If we can get it running in Verilator, then we can add it to the CI nightly regression.

davidharrishmc, 2026-11-07:
Reopening: testbench_fp now runs on Verilator after reworking its sequencing (one process applies and checks each vector, so the posedge races are gone). The full TestFloat regression (358 runs) passes on Verilator in about 10 minutes on a 32-core machine, including compilation. PR to follow.

The rework also found that the old `CheckNow` never checked f→int32 conversions, `lt` compares, or conversions to single or double, because their OpCtrl matched the div/sqrt encodings. Those vectors are now checked and pass.