openhwfoundation/cvw #715: No Definition Found on Verilator Compilation After Commit

Karl-Han, 2024-04-08:
Problem Statement

- When running regression on latest branch with both QuestaSIM and Verilator, QuestaSIM can pass the regressions without any problems while Verilator report the following Error Output.
- I checked out to `282f545d67d61219d7470db341b6a8fe97edcd54`, the commit right before the addition of MEMWb, and it turns out that the regression on Verilator works fine after patching the `7c3e93bb2cd4ffa2161aac28a92c34b649085d49` and `081cf5be55c40e33cd22e53947ae63b63c09a8b0`.

Command: 

- `cd ${CVW_HOME}/sim/verilator && verilator -GTEST="\"arch64i\"" -DVERILATOR=1 --timescale "1ns/1ns" --timing --binary --top-module testbench -I${CVW_HOME}/config/shared -I${CVW_HOME}/config/rv64gc ${CVW_HOME}/src/cvw.sv ${CVW_HOME}/testbench/testbench.sv ${CVW_HOME}/testbench/common/*.sv ${CVW_HOME}/src/*/*.sv ${CVW_HOME}/src/*/*/*.sv --relative-includes`
- CVE_HOME is the same as WALLY

Error Output:

```text
%Error: /home/cad/cvw/testbench/testbench.sv:585:40: Can't find definition of 'MEMWb' in dotted variabl
e: 'dut.uncore.uncore.uart.uart.MEMWb'                                                                 
  585 |       if (~dut.uncore.uncore.uart.uart.MEMWb & dut.uncore.uncore.uart.uart.u.A == 3'b000 & ~dut
.uncore.uncore.uart.uart.u.DLAB) begin                                                                 
      |                                        ^~~~~                                                   
%Error: /home/cad/cvw/testbench/testbench.sv:585:76: Can't find definition of 'u' in dotted scope/varia
ble: 'dut.uncore.uncore.uart.uart.u'               
  585 |       if (~dut.uncore.uncore.uart.uart.MEMWb & dut.uncore.uncore.uart.uart.u.A == 3'b000 & ~dut
.uncore.uncore.uart.uart.u.DLAB) begin                                                                 
      |                                                                            ^                   
        ... Known scopes under 'dut.uncore.uncore.uart.uart': <no instances found>                     
%Error: /home/cad/cvw/testbench/testbench.sv:585:121: Can't find definition of 'u' in dotted scope/vari
able: 'dut.uncore.uncore.uart.uart.u'              
  585 |       if (~dut.uncore.uncore.uart.uart.MEMWb & dut.uncore.uncore.uart.uart.u.A == 3'b000 & ~dut
.uncore.uncore.uart.uart.u.DLAB) begin             
      |                  
                         ^                                 ... Known scopes under 'dut.uncore.uncore.uart.uart': <no instances found>                     
%Error: /home/cad/cvw/testbench/testbench.sv:587:60: Can't find definition of 'u' in dotted scope/variable: 'dut.uncore.uncore.uart.uart.u'               
  587 |         $fwrite(memFile, "%c", dut.uncore.uncore.uart.uart.u.Din);                             
      |                                                            ^                                   
        ... Known scopes under 'dut.uncore.uncore.uart.uart': <no instances found>                     
%Error: Parameters from the command line were not found in the design: TEST                            
%Error: Exiting due to 5 error(s)                 
```

davidharrishmc, 2024-04-08:
This is now gated by P.UART_SUPPORTED at line 584.  Try pulling the latest version and see if that solves the Verilator issue.

davidharrishmc, 2024-04-16:
@Karl-Han, is this resolved by the PR?  If so, can you close the issue?

Karl-Han, 2024-04-18:
Yes. It is fixed by some PR, but I lost track of the number. Closing.

closed 2024-04-18 by Karl-Han