openhwfoundation/cvw #720: Zcb support needed in Sail

davidharrishmc, 2024-04-09:
gcc 13.2.0 now seems to support Zcb.

However, Sail does not yet seem to support it.  Running riscof on c.mul fails, hanging with an infinite trap loop.

more ../tests/riscof/riscof_work/rv32i_m/C/src/cmul-01.S/ref/cmul-01.log 
[78] [M]: 0x80000122 (0x6605) c.lui a2, 1
x12 <- 0x00001000
mem[X,0x80000124] -> 0x9E49
[79] [M]: 0x80000124 (0x9E49) c.illegal 40521
trapping from M to M to handle illegal-instruction

Here is a pending PR for Zcb into Sail.
https://github.com/riscv/sail-riscv/pull/322

Adding Zcb to Wally will inolve chanigng spike_rv{32/64}gc_isa.yaml, and regression-wally to have Zcb.
ISA: RV32IMAFDCZicsr_Zicond_Zifencei_Zfa_Zfh_**Zca_Zcb**_Zba_Zbb_Zbc_Zbkb_Zbkc_Zbkx_Zbs_Zknd_Zkne_Zknh
#

jordancarlin, 2024-04-16:
Zcb support was merged into Sail today: riscv/sail-riscv/pull/427

davidharrishmc, 2024-04-20:
Completed with PR747

closed 2024-04-20 by davidharrishmc