openhwfoundation/cvw #723: Accessing Environment Variable in rom1p1r.sv

europe2024, 2024-04-10:
Below is the platform.

     variant of Debian (version 12) on ARM
     Verilator 5.025 devel rev v5.024-2-g7ca2d6470

The following line

      $readmemh("$WALLY/fpga/src/boot.mem", ROM, 0);

in

     $WALLY/src/generic/mem/rom1p1r.sv

uses environment-variable $WALLY, but Verilator does not automatically generate code to get its value.

Below is one way to access $WALLY.

     rom1p1r.sv
     ----------
     import "DPI-C" function string getenv(input string env_name);

     $readmemh({getenv("WALLY"), "/fpga/src/boot.mem"}, ROM, 0);

Karl-Han, 2024-04-10:
Thanks for your advice. We are currently working on getting Verilator working in the team, and your insight to have C++ execution is great.

I will notify you when the feature is implemented and close the issue afterwards.

Karl-Han, 2024-04-12:
I tried to replicate your idea with passing RISCV in `testbench/testbench.sv`, but I cannot resolve the problem of undefined symbol. 

testbench/testbench.sv:
```systemverilog
import "DPI-C" function string getenvval(input string env_name);
string       RISCV_DIR = getenvval("RISCV"); // "/opt/riscv";
```

Log:
```shell
mold: error: undefined symbol: Vtestbench__ALL.a(Vtestbench___024unit__DepSet_hc0e221bc__0.o): getenvval
collect2: error: ld returned 1 exit status
make[1]: *** [Vtestbench.mk:66: Vtestbench_rv64gc] Error 1
```

Here are the questions I have:

- Can you share your command to compile the executable with Verilator?
- You are wrapping getenv with getenvval to avoid confliction in declaration of getenv in C and C++, right?

## Verilator Compilation Command

```shell
time verilator \
--Mdir obj_dir_non_profiling -o Vtestbench_rv64gc \
-cc --binary \
 -DVERILATOR=1 --no-trace-top --stats \
--timescale "1ns/1ns" --timing --top-module testbench  --relative-includes \
"-I/home/khan/cvw/config/shared" "-I/home/khan/cvw/config/rv64gc" \
wrapper.c \
/home/khan/cvw/src/cvw.sv /home/khan/cvw/testbench/testbench.sv /home/khan/cvw/testbench/common/*.sv /home/khan/cvw/src/*/*.sv /home/khan/cvw/src/*/*/*.sv
```

## wrapper.c
```c
#include <stdlib.h>

#include "Vtestbench.h"

const char *getenvval(const char *pszName) {
    return ((const char *) getenv(pszName));
}
```

europe2024, 2024-04-12:
In "wrapper.c", replace

     #include "Vtestbench.h"

with

     #include "Vtestbench__Dpi.h"

and Verilate the project again.

Karl-Han, 2024-04-12:
Thanks. It is very helpful as I am not familiar with C++ compilation of Verilator. I have submitted PR #728 .

davidharrishmc, 2024-04-16:
Is this resolved by PR728?  If so, can we close it?  If not, can @europe2024 or @Karl-Han make the necessary PR?

europe2024, 2024-04-17:
Karl-Han, do you plan to update the following file

         cvw/src/generic/mem/rom1p1r.sv

to use getenvval() to get the value of environment variable $WALLY?

europe2024, 2024-04-18:
@Karl-Han, a pull request requires cloning a fork of the project.  My Internet connection is too slow and faulty to support cloning.

Can you please do a pull request?

Earlier in the year, I used an alternative Internet connection to clone the project.  Using this clone, I ran the following command to test your proposed change.

     make -C sim/verilator WALLYCONF=rv32imc TEST=arch32i run

The change seems to work correctly.

Karl-Han, 2024-04-18:
Thanks for the confirmation. Opening PR for the patch and closing this issue.

closed 2024-04-20 by davidharrishmc