openhwfoundation/cvw #727: Limitation of Conditional Generate Construct

europe2024, 2024-04-12:
Verilator 5.025 devel rev v5.024-5-g1012c054e does not allow more than one block with the same name within a single conditional generate construct.

Until Verilator is updated to remove this limitation, multiple blocks with the same name in a conditional generate contruct must be assigned unique names.

See [https://github.com/verilator/verilator/issues/5046](url)

slmnemo, 2024-04-13:
Hello! I am working on fixing this. Do all blocks which share a name with their generate statement need to be renamed, and if so should we standardize the naming? Calling on Rose and Harris to help with renaming these blocks wherever they are. I have a preliminary renaming of just the UART and uncore that was causing Verilator to break in my most recent PR.

davidharrishmc, 2024-04-16:
@slmnemo is this fixed with your recent PR?  If not, is there a test case to show the problem?

europe2024, 2024-04-19:
@slmnemo, currently, the only known manifestation of the underlying problem in [#4769](https://github.com/verilator/verilator/issues/4769) occurs when you use a hierarchical path name to reference an object inside a generate block of a conditional generate construct. So, instead of uniquifying names of all such generate blocks, you could just uniquify the names of generate blocks related to each manifestation as it arises from the evolution of the SystemVerilog code for Wally.

A possible methodology for uniquifying the names is appending a string that no one else is likely to use.  Then, after the Verilator developers fix the underlying problem, you can easily find the places with your workaround and undo it.

closed 2024-04-22 by davidharrishmc