openhwfoundation/cvw #732: Encountering Error in Make While Simulating Core CVW with Questasim

WaseemEng, 2024-04-16:
[image]

I am simulating the core CVW .After following the steps outlined in the Git repository for setting up the CVW core simulation environment, I executed the source ./setup.sh command to ensure all configurations were appropriately adjusted, including path changes. However, upon running the make command to initiate the build process, I encountered the above error.

ross144, 2024-04-16:
Can you check that riscof is installed and in the PATH?
which riscof
/home/rose/.local/bin/riscof

If it's in the path can you run make in cvw/sim and report the error message?

davidharrishmc, 2024-04-17:
riscv-arch-test is out of date for you.  You'll need to pull a corrected version.

cd $WALLY/addins/riscv-arch-test
git pull

WaseemEng, 2024-04-17:
Thanks, @ross144, @quswarabid, and @davidharrishmc for your help in resolving the issue.

closed 2024-04-17 by WaseemEng