openhwfoundation/cvw #741: Questa and Verilator simulations do not match identically and give different CoreMark scores

davidharrishmc, 2024-04-17:
cd benchmarks/coremark
make clean
make
vsim rv32gc coremark
vsim rv32gc coremark -s verilator

**Questa results:**
#    WALLY CoreMark Results (from get_time)
#     Elapsed MTIME: 2972203
#     Elapsed MINSTRET: 2589000
#     COREMARK/MHz Score: 10,000,000 / 2972203 = 3.36
#     CPI: 2972203 / 2589000 = 1.14
# 2K performance run parameters for coremark.
# CoreMark Size    : 666
# Total ticks      : 2972203
# Total time (secs): 29
# Iterations/Sec   : 0
# Iterations       : 10
# Compiler version : GCC13.2.0
# Compiler flags   : -mcmodel=medany -fno-tree-loop-distribute-patterns -fno-common -lm -lgcc -T /home/harris/cvw/benchmarks/coremark/riscv64-baremetal/link.ld -g -mabi=ilp32 -march=rv32im_zicsr_zba_zbb_zbs -static -falign-functions=16 -mbranch-cost=1 -DSKIP_DEFAULT_MEMSET -mtune=sifive-3-series -O3 -finline-functions -falign-jumps=4 -fno-delete-null-pointer-checks -fno-rename-registers --param=loop-max-datarefs-for-datadeps=0 -funroll-all-loops --param=uninlined-function-insns=8 -fno-tree-vrp -fwrapv -fipa-pta -nostdlib -nostartfiles -ffreestanding -mstrict-align -DTOTAL_DATA_SIZE=2000 -DMAIN_HAS_NOARGC=1 -DPERFORMANCE_RUN=1 -DITERATIONS=10 -DXLEN=32   -static-libgcc -lgcc
# Memory location  : Code and Data in external RAM
# seedcrc          : 0xe9f5
# [0]crclist       : 0xe714
# [0]crcmatrix     : 0x1fd7
# [0]crcstate      : 0x8e3a
# [0]crcfinal      : 0xfcaf
# Correct operation validated. See README.md for run and reporting rules.
# Load Stalls 262774
# Store Stalls 26087
# D-Cache Accesses 698750
# D-Cache Misses 76
# I-Cache Accesses 2868917
# I-Cache Misses 246
# Branches 591846
# Branches Miss Predictions 66966
# BTB Misses 5570
# Jump and JR 31170
# RAS Wrong 45
# Returns 15244
# BP Class Wrong 19469

**Verilator results:**
WALLY CoreMark Results (from get_time)
    Elapsed MTIME: 2977225
    Elapsed MINSTRET: 2589000
    COREMARK/MHz Score: 10,000,000 / 2977225 = 3.35
    CPI: 2977225 / 2589000 = 1.14
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 2977225
Total time (secs): 29
Iterations/Sec   : 0
Iterations       : 10
Compiler version : GCC13.2.0
Compiler flags   : -mcmodel=medany -fno-tree-loop-distribute-patterns -fno-common -lm -lgcc -T /home/harris/cvw/benchmarks/coremark/riscv64-baremetal/link.ld -g -mabi=ilp32 -march=rv32im_zicsr_zba_zbb_zbs -static -falign-functions=16 -mbranch-cost=1 -DSKIP_DEFAULT_MEMSET -mtune=sifive-3-series -O3 -finline-functions -falign-jumps=4 -fno-delete-null-pointer-checks -fno-rename-registers --param=loop-max-datarefs-for-datadeps=0 -funroll-all-loops --param=uninlined-function-insns=8 -fno-tree-vrp -fwrapv -fipa-pta -nostdlib -nostartfiles -ffreestanding -mstrict-align -DTOTAL_DATA_SIZE=2000 -DMAIN_HAS_NOARGC=1 -DPERFORMANCE_RUN=1 -DITERATIONS=10 -DXLEN=32   -static-libgcc -lgcc
Memory location  : Code and Data in external RAM
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0xfcaf
Correct operation validated. See README.md for run and reporting rules.
Load Stalls 262721
Store Stalls 26248
D-Cache Accesses 698750
D-Cache Misses 76
I-Cache Accesses 2868812
I-Cache Misses 246
Branches 591811
Branches Miss Predictions 66966
BTB Misses 5570
Jump and JR 31170
RAS Wrong 45
Returns 15244
BP Class Wrong 19469
[5:35](https://hmce190ax.slack.com/archives/C01LED7HXNF/p1713353716135539)
[@Kunlin Han](https://hmce190ax.slack.com/team/U06MMB72X25)
 can you get the two to match identically?  I’m thinking something in the testbench is different for Verilator?

Karl-Han, 2024-04-18:
Here is what I have got from `make coremark`, which is run by questa
```text
#    WALLY CoreMark Results (from get_time)                           [24/1005]
#     Elapsed MTIME: 2972203                                                   
#     Elapsed MINSTRET: 2589000
#     COREMARK/MHz Score: 10,000,000 / 2972203 = 3.36 
#     CPI: 2972203 / 2589000 = 1.14
# 2K performance run parameters for coremark.
# CoreMark Size    : 666                                                       
# Total ticks      : 2972203
# Total time (secs): 29                                                        
# Iterations/Sec   : 0                                                         
# Iterations       : 10                                                        
# Compiler version : GCC13.2.0                                                 
# Compiler flags   : -mcmodel=medany -fno-tree-loop-distribute-patterns -fno-co
mmon -lm -lgcc -T /home/khan/cvw/benchmarks/coremark/riscv64-baremetal/link.ld 
-g -mabi=ilp32 -march=rv32im_zicsr_zba_zbb_zbs -static -falign-functions=16 -mb
ranch-cost=1 -DSKIP_DEFAULT_MEMSET -mtune=sifive-3-series -O3 -finline-function
s -falign-jumps=4 -fno-delete-null-pointer-checks -fno-rename-registers --param
=loop-max-datarefs-for-datadeps=0 -funroll-all-loops --param=uninlined-function
-insns=8 -fno-tree-vrp -fwrapv -fipa-pta -nostdlib -nostartfiles -ffreestanding
 -mstrict-align -DTOTAL_DATA_SIZE=2000 -DMAIN_HAS_NOARGC=1 -DPERFORMANCE_RUN=1 
-DITERATIONS=10 -DXLEN=32   -static-libgcc -lgcc
# Memory location  : Code and Data in external RAM
# seedcrc          : 0xe9f5                                                    
# [0]crclist       : 0xe714
# [0]crcmatrix     : 0x1fd7                                                    
# [0]crcstate      : 0x8e3a
# [0]crcfinal      : 0xfcaf
# Correct operation validated. See README.md for run and reporting rules.
# Load Stalls 262695
# Store Stalls 26086
# D-Cache Accesses 698748
# D-Cache Misses 75 
# I-Cache Accesses 2868636
# I-Cache Misses 246
# Branches 591764
# Branches Miss Predictions 66965
# BTB Misses 5570
# Jump and JR 31168 
# RAS Wrong 45
# Returns 15242
# BP Class Wrong 19469
```

The difference is as follows
[image]

Even though the ticks match between two versions of questa, the number of (actual/not retired) branches doesn't match, which means that branch predictors are working differently throughout the simulation, while the numbers of retired instructions, indicated by MINSTRET, are the same.

Therefore, I think this is related to the initialization and determinstic prediction of the branch predictor, because the numbers of retired instruction matches in these two questa version and verilator version.

## Casual Thought FYI
> The following is just some random thoughts without considering the meaning of MINSTRET. Ignore it if it doesn't make sense.

I wonder whether it is possible to get a vcs version for this. So we can compare them and get to know whether either one of them is not working properly. 

With the matching of vcs and questa, there are two possibilities:
- `$stop` and `$finish` are handled differently. Which causes the mismatches between vcs and verilator and between questa and verilator
- something goes wrong with verilator

As we compare the result side by side, we know that it is very likely that
- Verilator version executes a specific set of instructions more than questa, which leads to less load instructions, less ICache miss and less branches
- Verilator version does more store, which leads to increased ticks
[image]

Important questions are that 
- is it possible that the scheduling of event leads to different behaviors while preserving the same result?
- are we expecting the match in different simulators? 
- should we tolerate the differences in different simulators?

My answer to the above questions is that 
- all the behaviors should be the same
- there should be a match
- if there is differences, then it means that there are different machine code snippet that have been simulated.

Therefore, in the case of both Questa and Verilator simulations give the correct result while having different number of instructions, my intuition is that `$stop` and `$finish` is handled.

WaseemEng, 2024-04-24:
Here is what I have got from make coremark, which is run by questa.
[image]

closed 2024-06-18 by davidharrishmc