openhwfoundation/cvw #746: Wrong Address for "rvtest_entry_point" in "sumtest"

europe2024, 2024-04-18:
Below is the platform.

     variant of Debian (version 12) on ARM

"rvtest_entry_point" in "sumtest" must be at address 0x0000000080000000 so that the output of various tools match the textbook's examples involving "sumtest" in Chapter 3 RISC-V Software Tool Flow.

The files for compiling sumtest put "rvtest_entry_point" at 0x0000000080003000.

A possible solution is to change

     $WALLY/examples/link/link.ld

by replacing the following line

     .text : { *(.text.init) }

with the following line

     .text : { *(.text.init) *(.text) }

davidharrishmc, 2024-05-10:
Should be fixed with PR#791

davidharrishmc, 2024-05-13:
@europe2024 if you are satisfied, can you close the issue?

europe2024, 2024-05-14:
@davidharrishmc Thanks for updating the code.

closed 2024-05-14 by europe2024