openhwfoundation/cvw #768: Branch Predictor Performance Issue

ross144, 2024-05-02:
Changes since Dec 18, 2023 result in lower branch predictor accuracy by 1 to 2%.  We must resolve before TLR5 is completed.

ross144, 2024-08-20:
I am making notes here so I don't lose this data. I am currently unable to replicate this branch predictor issue.  I ran the branch embench benchmarks on commits from dec 18, 2023, jan 18, 2024, and feb 19, 2024 and there is only a tiny change in the accuracy between dec and jan < 0.01%.  My notes show I original found the issue in early feb 2024.  I am wondering if the issue is compiler related.  The programs I ran are compiled using the latest gcc 13.2.0.  I used the embench source code checked into the repo from dec 18, 2023 for each of the three simulations.

jordancarlin, 2024-08-20:
@ross144 if you wanted to test older compiler versions, there are downloadable, compiled binaries (not with quite the same multilib setup but should work) listed by date archived here: https://github.com/riscv-collab/riscv-gnu-toolchain/releases.

davidharrishmc, 2024-08-20:
How about we write this off as a fluke because it is working now, close the issue, and move on to the hotter issues.  We can return to these notes if we see it resurface.

closed 2024-09-12 by davidharrishmc