openhwfoundation/cvw #769: embench fails on chip.eng.hmc.edu

davidharrishmc, 2024-05-02:
```
$ cd $WALLY/benchmarks/embench
$ make clean
$ make run

...
wikisort
All benchmarks built successfully
../../addins/embench-iot/benchmark_size.py --builddir=bd_speedopt_size --json-output > wallySpeedOpt_size.json 
Traceback (most recent call last):
  File "/home/harris/cvw/benchmarks/embench/../../addins/embench-iot/benchmark_size.py", line 26, in <module>
    import lief
ModuleNotFoundError: No module named 'lief'
make: *** [Makefile:71: size] Error 1
```

slmnemo, 2024-05-03:
seems like pip install lief would fix this.

slmnemo, 2024-05-03:
seems sudo -H pip install lief should install it to the home directory of the system instead of the home directory of the user. try that?

davidharrishmc, 2024-05-03:
Closed with #776 per Kaitlin's -H approach.

closed 2024-05-03 by davidharrishmc