openhwfoundation/cvw #771: Remove flush from EBU critical path

davidharrishmc, 2024-05-02:
The EBU receives flush signals (necessary to solve a logic bug).  These are now on the critical path, but not in the version of the critical path reported in Chapter 6.

For example, buscachefsm has Flush affecting HTRANS. See the code and timing report below.  In the timing report, a branch misprediction raises FlushD, which affects HTRANS. busfsm is similar.

Rose Thompson feels the right fix is to issue the transaction anyway in the Execute stage, but to send a signal in the memory stage for the controller to cancel the transaction via the AHB protocol.  If this isn't feasible, it might be possible to kill HWSTRB in the memory stage so that transaction has no effect.

After the fix, rerun Linux boot in lock step, and regenerate Chapter 6 synthesis results.

```
  assign HTRANS = (CurrState == ADR_PHASE & HREADY & ((|BusRW) | (|CacheBusRW) | BusCMOZero) & ~Flush) |
                  (CurrState == ATOMIC_READ_DATA_PHASE) | 
                  (CacheAccess & FinalBeatCount & |CacheBusRW & HREADY & ~Flush) ? AHB_NONSEQ : // if we have a pipelined request
                  (CacheAccess & |BeatCount) ? (BURST_EN ? AHB_SEQ : AHB_NONSEQ) : AHB_IDLE;
```

```
PCSrcE (net)                                        0.005546            0.000000   2.174526 r
  ifu/PCSrcE (ifu__598949)                                                0.000000   2.174526 r
  ifu/PCSrcE (net)                                    0.005546            0.000000   2.174526 r
  ifu/U5/Y (sky130_osu_sc_12T_ms__buf_1)                        0.076231  0.110984   2.285510 r
  ifu/BPWrongE (net)                            2     0.007587            0.000000   2.285510 r
  ifu/BPWrongE (ifu__598949)                                              0.000000   2.285510 r
  BPWrongE (net)                                      0.007587            0.000000   2.285510 r
  hzu/BPWrongE (hazard__598949)                                           0.000000   2.285510 r
  hzu/BPWrongE (net)                                  0.007587            0.000000   2.285510 r
  hzu/U4/Y (sky130_osu_sc_12T_ms__inv_2)                        0.036381  0.048787   2.334297 f
  hzu/n8 (net)                                  3     0.009074            0.000000   2.334297 f
  hzu/U3/Y (sky130_osu_sc_12T_ms__inv_2)                        0.052211  0.057438   2.391736 r
  hzu/FlushD (net)                              4     0.010859            0.000000   2.391736 r
  hzu/FlushD (hazard__598949)                                             0.000000   2.391736 r
  FlushD (net)                                        0.010859            0.000000   2.391736 r
  ifu/FlushD (ifu__598949)                                                0.000000   2.391736 r
  ifu/FlushD (net)                                    0.010859            0.000000   2.391736 r
  ifu/bus.passthrough.ahbinterface/Flush (ahbinterface_32_0)              0.000000   2.391736 r
  ifu/bus.passthrough.ahbinterface/Flush (net)        0.010859            0.000000   2.391736 r
  ifu/bus.passthrough.ahbinterface/busfsm/Flush (busfsm_1)                0.000000   2.391736 r
  ifu/bus.passthrough.ahbinterface/busfsm/Flush (net) 0.010859            0.000000   2.391736 r
  ifu/bus.passthrough.ahbinterface/busfsm/U29/Y (sky130_osu_sc_12T_ms__nor2_1)
                                                                0.040983  0.046972   2.438707 f
  ifu/bus.passthrough.ahbinterface/busfsm/HTRANS[1] (net)
                                                2     0.004385            0.000000   2.438707 f
  ifu/bus.passthrough.ahbinterface/busfsm/HTRANS[1] (busfsm_1)            0.000000   2.438707 f
  ifu/bus.passthrough.ahbinterface/HTRANS[1] (net)    0.004385            0.000000   2.438707 f
  ifu/bus.passthrough.ahbinterface/HTRANS[1] (ahbinterface_32_0)          0.000000   2.438707 f
  ifu/IFUHTRANS[1] (net)                              0.004385            0.000000   2.438707 f
  ifu/IFUHTRANS[1] (ifu__598949)                                          0.000000   2.438707 f
  IFUHTRANS[1] (net)                                  0.004385            0.000000   2.438707 f
  ebu.ebu/IFUHTRANS[1] (ebu__598949)                                      0.000000   2.438707 f
  ebu.ebu/IFUHTRANS[1] (net)                          0.004385            0.000000   2.438707 f
  ebu.ebu/IFUInput/HTRANSIn[1] (controllerinput_PA_BITS34)                0.000000   2.438707 f
  ebu.ebu/IFUInput/HTRANSIn[1] (net)                  0.004385            0.000000   2.438707 f
  ebu.ebu/IFUInput/genblk1.RestorMux/d0[35] (mux2_WIDTH43)                0.000000   2.438707 f
  ebu.ebu/IFUInput/genblk1.RestorMux/d0[35] (net)     0.004385            0.000000   2.438707 f
  ebu.ebu/IFUInput/genblk1.RestorMux/U4/Y (sky130_osu_sc_12T_ms__oai21_l)
                                                                0.140213  0.116030   2.554738 r
  ebu.ebu/IFUInput/genblk1.RestorMux/n12 (net)
                                                1     0.004894            0.000000   2.554738 r
  ebu.ebu/IFUInput/genblk1.RestorMux/U3/Y (sky130_osu_sc_12T_ms__inv_2)
                                                                0.055106  0.071135   2.625873 f
  ebu.ebu/IFUInput/genblk1.RestorMux/y[35] (net)
                                                5     0.012306            0.000000   2.625873 f
  ebu.ebu/IFUInput/genblk1.RestorMux/y[35] (mux2_WIDTH43)                 0.000000   2.625873 f
  ebu.ebu/IFUInput/N0 (net)                           0.012306            0.000000   2.625873 f
  ebu.ebu/IFUInput/Request (controllerinput_PA_BITS34)                    0.000000   2.625873 f
  ebu.ebu/IFUReq (net)                                0.012306            0.000000   2.625873 f
  ebu.ebu/ebufsmarb/IFUReq (ebufsmarb)                                    0.000000   2.625873 f
  ebu.ebu/ebufsmarb/IFUReq (net)                      0.012306            0.000000   2.625873 f
  ebu.ebu/ebufsmarb/U16/Y (sky130_osu_sc_12T_ms__nand2_1)       0.057364  0.067131   2.693004 r
  ebu.ebu/ebufsmarb/n6 (net)                    1     0.004894            0.000000   2.693004 r
  ebu.ebu/ebufsmarb/U17/Y (sky130_osu_sc_12T_ms__inv_2)         0.039137  0.050475   2.743479 f
[... 19 lines cut ...]
  ebu.ebu/IFUInput/genblk1.SaveReg/U61/Y (sky130_osu_sc_12T_ms__nand2_l)
                                                                0.070212  0.062284   3.030937 f
  ebu.ebu/IFUInput/genblk1.SaveReg/n38 (net)
                                                1     0.002488            0.000000   3.030937 f
  ebu.ebu/IFUInput/genblk1.SaveReg/U31/Y (sky130_osu_sc_12T_ms__nand2_1)
                                                                0.048481  0.059047   3.089985 r
  ebu.ebu/IFUInput/genblk1.SaveReg/n46 (net)
                                                1     0.002852            0.000000   3.089985 r
  ebu.ebu/IFUInput/genblk1.SaveReg/q_reg[8]/D (sky130_osu_sc_12T_ms__dff_1)
                                                                0.048481  0.000000   3.089985 r
  data arrival time                                                                  3.089985

  clock clk (rise edge)                                                   2.525253   2.525253
  clock network delay (ideal)                                             0.000000   2.525253
  ebu.ebu/IFUInput/genblk1.SaveReg/q_reg[8]/CK (sky130_osu_sc_12T_ms__dff_1)
                                                                          0.000000   2.525253 r
  library setup time                                                      -0.135478  2.389775
  data required time                                                                 2.389775
  ----------------------------------------------------------------------------------------------
  data required time                                                                 2.389775
  data arrival time                                                                  -3.089985
  ----------------------------------------------------------------------------------------------
  slack (VIOLATED)                                                                   -0.700210

```

davidharrishmc, 2024-09-12:
Could be solved with an extra cycle of latency on bus requests.

rosethompson, 2024-10-10:
Resolved in a feature branch ahbopt.  Improved Arty A7 clock speed from 20 Mhz to 25 MHz. However, we are saving this for after tag 1.0 as this change won't match the book.

rosethompson, 2024-10-14:
The branch is ahbopt.

davidharrishmc, 2026-11-01:
Investigated the `ahbopt` implementation at `rosethompson/cvw@a163001c20ca987cabebfd1ea19aefe606a35542`, including matched controller synthesis and directed simulation. This still appears relevant, so I recommend keeping the issue open, but the old branch should not be merged unchanged.

### Launch penalty

The implementation adds an `IDLE` state before `ADR_PHASE`. A request arriving in IDLE does not drive HTRANS immediately; it is launched in the next cycle after surviving flush. This removes the direct combinational Flush → HTRANS/HWRITE/HBURST path.

With HREADY continuously high, directed controller tests measured:

| Operation | Original: first address / completion | ahbopt: first address / completion |
|---|---:|---:|
| Single uncached read through `buscachefsm` | cycle 0 / 2 | cycle 1 / 3 |
| Eight-beat cache fill | cycle 0 / 8 | cycle 1 / 9 |

Thus there is one extra initial cycle, not one extra cycle per beat. Cache hits do not incur this penalty. At unchanged clock frequency, bus-dependent work can get slower; a clock-frequency improvement must compensate. These are controller measurements, not a measured CoreMark or Linux performance delta.

### HREADY corner in the old branch

The branch changes the default next state in `ADR_PHASE` to `IDLE`, including when HREADY is low. That discards the pending address phase and retries through IDLE. Holding a read request asserted with HREADY alternating high/low exposes this repeating sequence:

| Cycle | HREADY | ahbopt current state | HTRANS |
|---|---:|---|---|
| 0 | 1 | IDLE | IDLE |
| 1 | 0 | ADR_PHASE | IDLE |
| 2 | 1 | IDLE | IDLE |
| 3 | 0 | ADR_PHASE | IDLE |
| … | … | … | … |

The original controller accepts the request at cycle 0; ahbopt accepts none during a 12-cycle directed test, and the state sequence repeats. This demonstrates a controller-level readiness corner; I have not established its reachability in the full core/interconnect. A port should hold ADR_PHASE while waiting for HREADY, with withdrawal/flush and arbitration behavior verified explicitly.

### Timing and area

Re-analysis of the saved RV64GC TSMC28 HPC+ TT 0.9 V, 25 C physical netlists exactly reproduced their existing timing:

| Implementation | Overall minimum period | Longest path through bus-FSM Flush → HTRANS |
|---|---:|---:|
| Original RV64GC | 1.774880 ns (563.4 MHz) | 1.691941 ns |
| AMO-read / late-word-selection experiment | 1.608300 ns (621.8 MHz) | 1.564827 ns |

The overall limiting paths are in the AMO/cache logic and do not pass through the changed bus-controller logic. Removing this flush path therefore does not directly remove the limiting path in these particular configurations. This does not invalidate the earlier reported Arty A7 benefit. I have not performed a full-core before/after synthesis of ahbopt; placement and buffering could change the final result.

Matched topographical synthesis of the instruction/data controller pair, with the instruction-side unused request inputs tied off, measured **102.816 → 112.518 µm²**, or **+9.702 µm²**, and **17 → 18 sequential cells**. This used eight-beat lines, bursts enabled, a 1.1 ns target, and identical loads and a 100 × 100 µm placement region. The increment is about 0.005% of the reference core's non-RAM cell area as a scale comparison, not a measured whole-core area delta. No RAM or wide address/data pipeline registers are added.

The branch only changes `buscachefsm.sv`; the separate `busfsm.sv` passthrough controller still has the flush path shown in this issue's original timing excerpt. Together with the HREADY corner and the outstanding full-core regression/Linux lockstep validation, that is why I would leave this issue open.

closed 2026-11-06 by davidharrishmc