openhwfoundation/cvw #779: More tests in nightly regression

davidharrishmc, 2024-05-06:
Performance validation: CoreMark, Embench, cache, branch, ways to detect issues
Tests with multiple simulation
Lockstep tests
Code coverage
Functional coverage

Some of this depends on having a single test bench that supports lockstep, and replacing run-elf / run-elf-cov, iter-elf.

Update testplan.md when done

davidharrishmc, 2024-05-13:
VCS added to nightly regression.

davidharrishmc, 2024-07-06:
Added lockstep tests of coverage and wally-riscv-arch-test privilege with iterelf PR #866 
Added lockstep test of buildroot boot

jordancarlin, 2024-11-21:
Might be time to start thinking about adding functional coverage from cvw-arch-verif to nightly regression

jordancarlin, 2024-12-31:
Also add riscv-arch-test priv tests in lockstep to nightly. This should include `privileged`, `pmp*`, and `vm_sv*`.

jordancarlin, 2025-05-13:
Coremark, embench, and cache validation added to nightly.

Looks like the only things left from this issue are branch prediction and maybe checking code/functional coverage in nightly.

rosethompson, 2026-11-06:
Cannot run lockstep on CI because it must be run locally.