openhwfoundation/cvw #786: wsim is not running testbench_fp in Verilator

davidharrishmc, 2024-05-08:
```
wsim fdqh_ieee_rv64gc add -t testbench_fp

works in Questa but

wsim fdqh_ieee_rv64gc add -t testbench_fp --sim verilator

is not passing the TEST properly

- V e r i l a t i o n   R e p o r t: Verilator 5.025 devel rev v5.024-12-ga9664b8d1
- Verilator: Built from 310.531 MB sources in 244 modules, into 81.514 MB in 109 C++ files needing 114.560 MB
- Verilator: Walltime 5.272 s (elab=0.209, cvt=2.771, bld=1.782); cpu 0.000 s on 1 threads; alloced 346.250 MB
mkdir -p /home/harris/cvw/sim/verilator/logs
wkdir/fdqh_ieee_rv64gc_add/Vtestbench +TEST=add
TEST add not supported in this configuration
- /home/harris/cvw/testbench/testbench.sv:223: Verilog $finish
- S i m u l a t i o n   R e p o r t: Verilator 5.025 devel
- Verilator: $finish at 5ps; walltime 0.003 s; speed 0.000 s/s
- Verilator: cpu 0.000 s on 1 threads; alloced 7367 MB
```

davidharrishmc, 2024-05-10:
Fixed passing TEST with 791, but Verilator still is not running TestFloat correctly.

stineje, 2024-05-10:
This could be the way I made some of the changes in testfloat-fp.sv several months ago.  I can look at it tomorrow and see if I can fix.  Specifically, I am concerned my repeat items in the testbench-fp.sv are not compatible with ventilator.  I have them on my list to fix.  I am happy to look tomorrow.  Apologies if it’s due to my original issues.

ross144, 2024-05-14:
This should be fixed with PR #802

davidharrishmc, 2024-05-15:
Verilator now runs the test.  It does not produce the correct results that Questa produces, but that is a separate known issue.  Closing the issue.

closed 2024-05-15 by davidharrishmc