openhwfoundation/cvw #795: Can ground exact result be optimized?

davidharrishmc, 2024-05-13:
If Exact is true, can we make sure the fround logic always produces the right answer without the resultmux in fround.sv

jordancarlin, 2024-05-26:
Eliminating `Exact` would also allow for `EgeNf`. The current implementation does not produce the correct result without `Exact`, but is seems likely that with some small tweaks it would be possible.

davidharrishmc, 2024-06-10:
I hoped that
assign IMask = {1'b1, {P.NF{1'b0}}} >>> E;
would create all 1s when E >= Nf.

However, on result 25 of D_zfa/src/fround_b1-01.S, this doesn't work.
The case involves rounding the largest single-precision nan-boxed normal number (deadbeef7f7fffff)
E = 127 (max normal) and Xm = 1fff...ffe00.000 
But Imax = 0, rather than all 1's.

Apparently Questa SystemVerilog produces 0 when arithmetic right shifting by more than the word size. This seems to diagree with section 11.4.10 of the 1800-2017 spec:
"It shall fill the vacated bit positions with the value of the most significant (i.e., sign) bit of the left operand if the result type is signed. "

In any event, Imask is not right for this large exponent, so we won't be able to reuse the ordinary data path for exact results.  

I can still optimize out the Zero check, but will leave EgeNf.