openhwfoundation/cvw #828: Issue with zknde32 K extension regression

stineje, 2024-06-07:
There seems to be an error.  The shamt on the sbox does not convert correctly when its look at the state from a byte.  The original code is on L42 of zknde32.sv

assign shamt = {bs, 3'b0};

The following will fix this error correctly

assign shamt = {3'b0, bs} << 3;

This will handle the shamt and allow the index select part to allow the conversion to be multiplied by x8.  I am currently doing the PR for the JTAG so cannot put in at the moment.  I am just putting this item in once I get the JTAG in or someone else can add it.  Running regression-wally with this changer gives 0 errors.

davidharrishmc, 2024-06-07:
The zknde tests pass quests. What tool is flagging an error?

stineje, 2024-06-07:
Mine does not and I verified it with a fresh test and pull .  Can post shortly.

stineje, 2024-06-07:
I verified with Kelvin and he helped me figure it out.  I believe Rose also saw the error.  We are using the following hash:  5dfde808f003bf0eb94a5627b5c1a0a40128d347

wsim --sim questa  rv32gc arch32zkne > /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zkne.log: Failures detected in output
  Check /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zkne.log wsim --sim questa  rv32gc arch32zknd > /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zknd.log: Failures detected in output   Check /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zknd.log

I just tried again and get the same error. I sam confused why you have no error.

# Loading work.cacheway(fast__1)
#   Error on test rv32i_m/K/src/aes32esi-01.S result           3: adr = 8000411c sim (D$) d069e102 signature = de69
e102
# ** Note: $stop    : /home/jstine/cvw/testbench/testbench.sv(918)

#   Error on test rv32i_m/K/src/aes32dsi-01.S result           3: adr = 8000411c sim (D$) e169e102 signature = f869 e102 # ** Note: $stop    : /home/jstine/cvw/testbench/testbench.sv(918)

I will try again to validate

stineje, 2024-06-07:
I tried again and got the same error (5dfde808f003bf0eb94a5627b5c1a0a40128d347).  Maybe it’s something I did but I think the error seems valid.

lint-wally  | tee /home/jstine/cvw/sim/verilator/logs/all_lints.log: Success
wsim --sim questa  rv32gc arch32zknd > /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zknd.log: Failures detected in output
  Check /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zknd.log
wsim --sim questa  rv32gc wally32periph > /home/jstine/cvw/sim/questa/logs/rv32gc_wally32periph.log: Success
wsim --sim questa  rv32gc arch32zbkc > /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zbkc.log: Success
wsim --sim questa  rv32gc arch32zkne > /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zkne.log: Failures detected in output
  Check /home/jstine/cvw/sim/questa/logs/rv32gc_arch32zkne.log

davidharrishmc, 2024-06-08:
It's working for me. See below for versions.

I also don't see how your proposed fix changes anything. bs is 2 bits.  shamt is 5 bits.  
    assign shamt = {bs, 3'b0};          // shamt = bs * 8 (convert bytes to bits)
is a perfectly correct way to put bs in the top 2 bits of the 5-bit bus.  

I presume you've pulled the latest version?

```
harris@chips:~/cvw$ wsim --sim questa  rv32gc arch32zkne
...
# rv32i_m/K/src/aes32esi-01.S succeeded.  Brilliant!!!
# rv32i_m/K/src/aes32esmi-01.S succeeded.  Brilliant!!!
# SUCCESS! All tests ran without failures.
# ** Note: $stop    : /home/harris/cvw/testbench/testbench.sv(455)
#    Time: 71820 ns  Iteration: 1  Instance: /testbench
# Break at /home/harris/cvw/testbench/testbench.sv line 455
# Stopped at /home/harris/cvw/testbench/testbench.sv line 455
# End time: 23:44:49 on Jun 07,2024, Elapsed time: 0:00:01
# Errors: 0, Warnings: 0
harris@chips:~/cvw$ git log
commit 5dfde808f003bf0eb94a5627b5c1a0a40128d347 (HEAD -> dev, upstream/main)
harris@chips:~/cvw$ which vsim
/cad/mentor/questa_sim-2023.4/questasim/bin/vsim
```

stineje, 2024-06-08:
I agree with you, but I tried on 3 more machines here and got the same error.  Yes, the fix is the same but something is happening here.  I even tried regenerating the tool chain on one machine and still got the same error.  This does not make sense as the nightly regression doesn’t show anything.  I will look at it Monday - working on getting JTAG all cleared and tested.   

I still think it’s worthy having this as an issue until we can figure out what is going on.  My guess is something with Questa.

stineje, 2024-06-19:
Issue is fixed in Quests 2023.3_2 and onward.   No reference in the release information what was fixed that pertained to this problem.  Nevertheless, the problem is fixed in the current release.

closed 2024-06-19 by stineje