openhwfoundation/cvw #832: wsim does not properly pass argument to Verilator

davidharrishmc, 2024-06-14:
```
harris@chips:~/cvw$ wsim --sim verilator  buildroot buildroot --args +INSTR_LIMIT=1400000
Config=buildroot tests=buildroot sim=verilator gui=False args='+INSTR_LIMIT=1400000'
[]
Running Verilator on buildroot buildroot
make: Entering directory '/home/harris/cvw/sim/verilator'
mkdir -p wkdir/buildroot_buildroot
verilator \
--Mdir wkdir/buildroot_buildroot -o Vtestbench \
--binary --trace \
 -DVERILATOR=1 --no-trace-top --stats \
+INSTR_LIMIT=1400000 \
--top-module testbench  --relative-includes \
"-I/home/harris/cvw/config/shared" "-I/home/harris/cvw/config/buildroot" "-I/home/harris/cvw/config/deriv/buildroot" \
/home/harris/cvw/sim/verilator/wrapper.c \
 \
/home/harris/cvw/src/cvw.sv /home/harris/cvw/testbench/testbench.sv /home/harris/cvw/testbench/common/*.sv /home/harris/cvw/src/*/*.sv /home/harris/cvw/src/*/*/*.sv
%Error: Invalid option: +INSTR_LIMIT=1400000
make: *** [Makefile:60: wkdir/buildroot_buildroot/Vtestbench] Error 1
make: Leaving directory '/home/harris/cvw/sim/verilator'
```

jordancarlin, 2024-06-28:
Seems like this is resolved. On the latest version of the repo, Verilator passes the buildroot test and stops at the appropriate instruction.

```
jcarlin@chips:~/cvw$ wsim --sim verilator  buildroot buildroot --args +INSTR_LIMIT=1400000
Config=buildroot tests=buildroot sim=verilator gui=False args='+INSTR_LIMIT=1400000'
[]
Running Verilator on buildroot buildroot
make: Entering directory '/home/jcarlin/cvw/sim/verilator'
mkdir -p /home/jcarlin/cvw/sim/verilator/logs
wkdir/buildroot_buildroot/Vtestbench +TEST=buildroot +INSTR_LIMIT=1400000 
Reached               100000 instructions
Reached               200000 instructions
Reached               300000 instructions
Reached               400000 instructions
Reached               500000 instructions
Reached               600000 instructions
Reached               700000 instructions
Reached               800000 instructions
Reached               900000 instructions
Reached              1000000 instructions
Reached              1100000 instructions
Reached              1200000 instructions

OpenSBI v1.3
   ____       Reached              1300000 instructions
             _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|___/_____|
        | |
        |_|

Reached              1400000 instructions
- /home/jcarlin/cvw/testbench/testbench.sv:666: Verilog $finish
- S i m u l a t i o n   R e p o r t: Verilator 5.025 devel
- Verilator: $finish at 15us; walltime 14.222 s; speed 0.000 s/s
- Verilator: cpu 0.000 s on 1 threads; alloced 826 MB
make: Leaving directory '/home/jcarlin/cvw/sim/verilator'
```

davidharrishmc, 2024-06-28:
Agreed.

closed 2024-06-28 by davidharrishmc