openhwfoundation/cvw #839: IgnoreRequest logic in IFU and LSU

ross144, 2024-06-19:
Think through the purpose of IgnoreRequest in the IFU and LSU.  It is unused in the IFU and only use by atomic in the LSU.  IgnorerRequestTLB is generated by the HPTW to assert HPTWFaults and TLB misses.

ross144, 2024-06-19:
This is the unused code from the IFU. Lines 214 to 215.

```
  logic  IgnoreRequest; // *** unused; RT: is this a bug or delete?
  assign IgnoreRequest = ITLBMissF | FlushD;
```

ross144, 2024-06-19:
In lrsc.sv Line 48.

`  // possible bug: *** double check if PreLSURWM needs to be flushed by ignorerequest.`

rosethompson, 2024-10-11:
IgnoreRequest's purpose is to flush the cache/bus/dtim request as if it was a pipeline FlushW when the I/DTLB misses and the HPTW makes requests to the LSU.  It also checks for non-zero PBMT bits in non-leaf PTE during a walk and should suppress the memory request and generate a fault.  As I've simplified the code I've uncovered a bug.  I don't think we are actually checking the non-leaf PTE in the correct cycle.  We check for a HPTWFault in the L3_RD, L2_RD, L1_RD, and L0_RD states which is fine for page faults, access faults, etc because the read access occurs in the same cycle. However we register ReadData into the PTE register and then check the PBMT bits for a fault.  This means the FSM needs to check the L3_ADR, L2_ADR, L1_ADR, and L0_ADR states also.  Alternatively the PBMT fault check could occur directly from ReadData.

I want to do some more investigating.  I've already simplified the code a decent amount and it make more sense overall.

rosethompson, 2024-10-13:
Pull request #1004 resolves my questions about IgnoreRequest.

davidharrishmc, 2024-10-14:
Closed with PR #1004

closed 2024-10-14 by davidharrishmc