openhwfoundation/cvw #840: FPU fails when supervisor mode is not supported

jordancarlin, 2024-06-19:
When `S_SUPPORTED` is set to 0 and `F_SUPPORTED` is set to 1, the archXXf tests fail and get stuck in a loop of illegal instructions (as seen below)

`Program fetched illegal instruction 0x00000000 from address 0x00000000.  Might be fault with no fault handler.`

Floating point should not be dependent on privilege modes being supported, just Zicsr. None of the current derived configurations test with s/u modes disabled (other than the ones based on simpler configs like rv32imc), which is likely why this has not been discovered.

jordancarlin, 2024-06-19:
To reproduce, modify the rv32gc or rv64gc config by setting `S_SUPPORTED = 0`. When `S_SUPPORTED` is turned off, `SSTC_SUPPORTED` and `VIRTMEN_SUPPORTED` (along with the additional virtual memory extensions) must also be turned off.

davidharrishmc, 2024-06-19:
Great find!  Can you create a derived config to test this?  And do you want to search for the root cause and a fix?

jordancarlin, 2024-06-19:
I'll definitely create a derived config for it. Should probably also do derived configs for a bunch of the other features with privilege modes disabled. I can also look into a root cause/fix. Nothing immediately obvious, but hopefully something will jump out.

PR #855 (2024-06-27): FPU without privilege modes + derived config fixes
- Fix derived configs with `D_SUPPORTED` after #853 stopped `ZCD_SUPPORTED` from being turned off
- Add derived configs for no privilege modes
- Fix #840 FPU failing when `S_SUPPORTED = 0` by adjusting logic for updating `STATUS_FS`
closed 2024-06-27 by davidharrishmc