openhwfoundation/cvw #843: HPTW accesses not checked by PMA/P checkers

ross144, 2024-06-19:
Each memory access in a HPTWalk needs to be potentially flushed if the PMA/P checkers
 generate an access fault.  Specially the store on UDPATE_PTE needs to check for access violation.
  // I think the solution is to do 1 of the following
  // 1. Allow the HPTW to generate exceptions and stop walking immediately.
  // 2. If the store would generate an exception don't store to dcache but still write the TLB.  When we go back
  // to LEAF then the PMA/P.  Wait this does not work.  The PMA/P won't be looking a the address in the table, but
  // rather than physical address of the translated instruction/data.  So we must generate the exception.

Line 272 of hptw.sv

Logic bug not caught by regression tests.

ross144, 2024-06-19:
This was fixed a while ago, but there isn't a test to show that is works.

rosethompson, 2024-10-01:
Added tests with pull request #981

closed 2024-10-01 by rosethompson