openhwfoundation/cvw #845: Concurrent HPTW access fault with IFU bus fetch

ross144, 2024-06-19:
Issue #412 exposed a bug in Wally with a concurrent I$ miss and DTLB miss.  The HPTW makes a request which generates an access fault during an inflight I$ bus request.  This Trapped during a bus access violating the atomic nature of the bus request.  
A simulation solution was to suppress the fault in trap.sv...

`  assign TrapM = (ExceptionM & ~CommittedF) | InterruptM; // *** RT: review this additional ~CommittedF with DH and update priv chapter.` 

However this creates the potential hazard of missing exceptions.  This warrants careful review and a better solution.

Ideally we allow the exception and find a way to either delay it or abort the ifu transaction. 

Probably need to record the exception during the walk.  Abort the walk.  Wait for the ifu to finish, then replay the exception and take the trap.

Could also think more deeply about the entire exception architecture.

rosethompson, 2024-10-14:
Confirmed this is a bug.  I extended the hptwAccessFault.S test to create the specific condition above and the HPTW walks the page table multiple times while the IFU stalls the CPU.  Fortunately it still produces the same exception, but it is wasting power and cycles.

rosethompson, 2024-10-14:
I have several thoughts about how to fix this. 

1. Easy.  We an ignore since the bug is only performance.
2. Simple. Use CommittedF to keep the hptw FSM in the Fault state.  Some extra work is required to keep asserting the exception.  I don't like that CommittedF or StallF now has to come into the hptw.
3. Hard. The underlying problem is the IFU is performing a bus read while the hptw running. If it generates an exception, PCNextF needs to immediately point to the trap handler.  We could decouple the bus controller from PCF/PCNextF.  This would allow us to update PCF on a trap and still finish the bus transaction.  We'd have to keep holding the CPU stalled during the trap which requires changing some PC muxes.  Another advantage of this version is it would allow us to remove CommittedF from the TrapM logic for both exceptions and interrupts.

I think for now we should ignore the problem since it's just a performance bug, but do option 3 as we build the Wally enhancements.

davidharrishmc, 2026-11-01:
@rosethompson, I looked into this, and the `~CommittedF` term doesn't lose exceptions. Draft PR #2042 is for your review.

- `CommittedF` implies `IFUStallF`, so while an exception is deferred the faulting instruction stays in M with the same PC, cause and tval.
- The instruction's own access is cancelled (`SelfFaultM`, #1986), and a misaligned access does not start its second half.
- Only the LSU can raise an exception after an IFU fetch commits: a page table walk fault, a fault found after the walk fills the DTLB, or a fault on the second half of a misaligned access.

To check this, a per-cycle monitor ran over the coverage tests, the rv64gc arch tests, and a new directed test, `deferredwalk.S`. It found no lost or altered exceptions. `deferredwalk` also runs in lockstep with no mismatches over 1187 traps, and it fails if `~CommittedF` is removed.

The only defect is a performance one. When a walk faulted during an I$ fill, the walker's one-cycle fault pulse was dropped and the walk repeated (about 5 cycles). The PR keeps the walker in FAULT until `CommittedF` falls, at negligible cost. If you agree, the PR closes this issue.

closed 2026-11-06 by rosethompson