openhwfoundation/cvw #857: Issues with wsim

ross144, 2024-06-27:
1. Elf mode requires running from sim/questa
2. Directory mode does not quit cleanly on ctrl-c.  Just starts the next simulation.
3. On directory print a summary and put details into a log file.  Wrap wsim like iter-elf so that is produces success or failure for each elf file.

ross144, 2024-06-27:
4. The work directory gets named by the path of the elf file or the directory.  Need a better naming scheme.

ross144, 2024-06-27:
5. functional coverage cov/rv64gc_none.ucdb does not contain any functional coverage after running wsim with --fcov

davidharrishmc, 2024-07-03:
Elf mode seems to run when the full path to the ELF is provided.
wsim rv64gc $WALLY/tests/riscof/work/riscv-arch-test/rv64i_m/I/src/addi-01.S/ref/ref.elf --lockstep --sim questa

davidharrishmc, 2024-07-03:
Solved with PR #866 to replace wsim directory mode with iterelf that runs jobs in parallel.
Solved the elf path issue by adding os.path.abspath to wsim

closed 2024-07-03 by davidharrishmc