openhwfoundation/cvw #864: Wally mismatch with ImperasDV

davidharrishmc, 2024-07-03:
The MISA B bit is set in Wally, but not in the 20240305 release of ImperasDV.  Queried Lee about how to fix.

wsim buildroot buildroot --lockstep --args +INSTR_LIMIT=1400000

```
 Info (IDV) Instruction executed prior to mismatch '0x80004776(): 301027f3 csrr    x15,misa'
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000000080004776 ):
# Error (IDV) Mismatch 0> GPR x15
# Error (IDV)   . dut:0x800000000014112f
# Error (IDV)   . ref:0x800000000014112d
```

davidharrishmc, 2024-07-03:
@ross144 thinks we need the newest ImperasDV.

eroom1966, 2024-07-04:
Hi David,
looks like I need some pre-requisites, firstly, I tried running under VCS with this command line
wsim buildroot buildroot --lockstep --args +INSTR_LIMIT=1400000 --sim vcs

eroom1966, 2024-07-04:
I was able to reproduce this with a simpler testcase.
As we discussed the misa[B] bit has flip/flopped between present, not present in different releases
In the current release, it is now optional, so there is a parameter (misa_B_Zba_Zbb_Zbs) to control this

I added the following to your imperas.ic file 

```
# bit manipulation
--override cpu/add_Extensions=B 
#--override cpu/add_implicit_Extensions=B 
--override cpu/bitmanip_version=1.0.0
--override cpu/misa_B_Zba_Zbb_Zbs=T
```

This will fix the issue, but you may need an update to your simulator, I am not sure what version you are running

davidharrishmc, 2024-07-04:
Fixed with ImperasDV 20240703

closed 2024-07-04 by davidharrishmc