openhwfoundation/cvw #867: Many custom tests are failing in VCS lockstep but passing in Questa lockstep

davidharrishmc, 2024-07-03:
Some custom tests completely pass in lockstep with both VCS and Questa.  For example:

iterelf tests/riscof/work/riscv-arch-test/rv64i_m/I/src --sim vcs

completely works.

However, 
iterelf tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src --sim vcs
has 16/38 failures, while with questa, it only has 2/38 failures.

Similarly,
iterelf tests/coverage --sim questa
has 3/28 failures (vm64check, tlbmisc, fpu), while vcs has 20/28 failures

eroom1966, 2024-07-04:
I have done the following

# riscof
sudo -H pip3 install -U testresources riscv_config
sudo -H pip3 install git+https://github.com/riscv/riscof.git

cd cvw/tests/riscof
make

but I get a failure
   ERROR | Error in /home/moore/git/eroom1966/openhwgroup/cvw/cvw/tests/riscof/spike/spike_rv32gc_isa.yaml.
   ERROR |  ISA:
   ERROR |   - Input ISA string does not match accepted canonical ordering
   ERROR |   - Input ISA string does not match regex

jordancarlin, 2024-07-04:
@eroom1966 I've seen that issue come up when the version of riscv-config is not up to date. Try running `riscv-config --version`. It should be Version: 3.18.2. If it isn't, you should be able to update by doing `sudo -H pip3 -U riscv_config`. 

Also, did you do `source setup.sh` from the main cvw directory?

davidharrishmc, 2024-07-04:
Fixed with Imperas 20240703 and PR #866

closed 2024-07-04 by davidharrishmc

eroom1966, 2024-07-08:
Hi @jordancarlin 

> @eroom1966 I've seen that issue come up when the version of riscv-config is not up to date. Try running `riscv-config --version`. It should be Version: 3.18.2. If it isn't, you should be able to [...]

If I run 'riscv-config --version', I get 
```
riscv-config --version
RISCV-CONFIG: RISC-V Configuration Validator
Version: 3.7.2
```

If I try 'sudo -H pip3 -U riscv_config', I get
```
Usage:   
  pip <command> [options]
no such option: -U
```

in the instrucions I see the following in ./bin/wally-tool-chain-install.sh
```
# riscof
sudo -H pip3 install -U testresources riscv_config
sudo -H pip3 install git+https://github.com/riscv/riscof.git
```
Both commands complete without error

jordancarlin, 2024-07-08:
My mistake. It should have been `sudo -H pip3 install -U riscv_config`. If you ran the line from the tool chain install again and it worked, that should have done the same thing though. 

It’s possible the version of pip is too old. Try running the modified command above to just upgrade riscv_config and see if it says something about not updating due to an old version of something else (could be pip or some other dependency). If the issue is with pip, you can upgrade it by running `pip install --upgrade pip`

eroom1966, 2024-07-08:
@jordancarlin 
the two commands for riscof succeeded but I am not seeing the version of riscv_config you refer to.
when running 'pip install --upgrade pip'

```
Command 'pip' not found, but can be installed with:
apt install python-pip
```

So I installed, then went back to run commands, but I still get
```
riscv-config --version
RISCV-CONFIG: RISC-V Configuration Validator
Version: 3.7.2
```
each time I rerun the pip3 install - for all the packages it tells me "Requirement already satisfied:..."
I am at a loss regarding the problem here

jordancarlin, 2024-07-08:
What version of python (`python --version` or `python3 --version`) are you running? The newest version of riscv-config requires at least 3.7.0. And which version of Ubuntu?

eroom1966, 2024-07-08:
I think this is my problem
```
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"

$ python3 --version
Python 3.6.9
```

I have access to a 22.04 machine, it seems to be working on that ....

jordancarlin, 2024-07-08:
Yes. I have been working on an updated installation script and have things working as far back as Ubuntu 20.04, but not 18.04. You'd need to get a newer version of python (and likely newer gcc for some of the other tools) installed to use 18.04.

eroom1966, 2024-07-09:
@jordancarlin 
Lets take these problems onto a different issue
https://github.com/openhwgroup/cvw/issues/876