openhwfoundation/cvw #871: Restore verbose mode for ImperasDV

davidharrishmc, 2024-07-05:
We've lost the ability to run ImperasDV in verbose mode now that we transitioned to wsim.

It seems that we used to do it by passing OTHERFLAGS= *** to vsim in wally.do.

Some scripts such as setup.imperas.sh or sim-imperas set these as

OTHERFLAGS="+TRACE2LOG_ENABLE=1 VERBOSE=1"

Adding 
--trace
--tracemode
to imperas.ic gives us the tracing back, but it would be good to be a wsim option.

jordancarlin, 2024-08-15:
@davidharrishmc With the changes made in #908, I think we can get this working again. Would you prefer a single option that you pass to wsim that enables a specific set of options, a flag that you pass to wsim that lets you specify desired options (in line with something like --args or the new --params), or using the value of the OTHERFLAGS environment variable?

davidharrishmc, 2024-08-15:
The wsim user shouldn’t have to remember how to type a bunch of stuff, so it should just be a command line flag.

Maybe --lockstep --verbose.  But maybe --lockstepverbose is better because verbose only modifies lockstep and wouldn’t be used on its own.

PR #941 (2024-09-10): Add Imperas lockstep verbose as wsim option
Adds a `--lockstepverbose` flag to wsim that runs lockstep in verbose mode. `--lockstepverbose` automatically implies `--lockstep`. It's a bit hacky and uses a second `imperas-verbose.ic` file that is added to the main `imperas.ic` file when `--lockstepverbose` is passed, but was the cleanest solution I could find.

Fixes #871
closed 2024-09-10 by ross144