openhwfoundation/cvw #881: ImperasDV RV64GCK lockstep mismatch because ImperasDV considers grevi to be legal

davidharrishmc, 2024-07-14:
coverage/ieu.elf matches ImperasDV with the RV64GC model, but not with RV64GCK.

Change IDV_CONFIG_MODEL_VARIANT to RV64GCK in testbench/testbench.sv, then invoke with the following command.  The simulator doesn't matter.

wsim rv64gc dummy --elf /home/harris/cvw/tests/coverage/ieu.elf --sim vcs --lockstep

Lockstep sim will fail with a mismatch at 8000306a, which is a grevi instruction.  This instruction has not been ratified.  I think ImperasDV should throw an Illegal Instruction exception to match the Wally DUT.

[ieu.tar.gz](https://github.com/user-attachments/files/16215742/ieu.tar.gz)

```
Info 390: 'refRoot/cpu', 0x000000008000306a(main+6a): Machine 6bf05013 grevi   x0,x0,63
Info   MEMX 0x8000306a 0x8000306a 2 5013
Info   MEMX 0x8000306c 0x8000306c 2 6bf0
Info (IDV) Instruction executed prior to mismatch '0x800000e6(updateepc+16): 30200073 mret'
Error (IDV) PC mismatch (HartId:0, PC:0x000000008000306a main+6a):
Error (IDV) Mismatch 0>
Error (IDV)   . dut:0x0000000080000050 trap_handler+0
Error (IDV)   . ref:0x000000008000306a main+6a
```

davidharrishmc, 2024-07-14:
Once this is fixed, verify coverage runs with

iterelf tests/coverage

eroom1966, 2024-07-15:
Hi David - if I run this test I get no miscompares, and the instruction does cause an exception, and goes off o the trap handler
Please remind me the version of Imperas you are using so I can double check against that particular version

```
Info 390: 'refRoot/cpu', 0x000000008000306a(main+6a): Machine 6bf05013 grevi   x0,x0,63
Info 390: 'refRoot/cpu',   MEMX 0x8000306a 0x8000306a 2 5013
Info 390: 'refRoot/cpu',   MEMX 0x8000306c 0x8000306c 2 6bf0
Warning (RISCV_ILL) CPU 'refRoot/cpu' 0x8000306a 6bf05013 grevi   x0,x0,63: Illegal instruction - Zbp absent
Info 390: 'refRoot/cpu',   mstatus 0000000a00000088 -> 0000000a00001880 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0->3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:1->0 SIE:0]
Info 390: 'refRoot/cpu',   mtval 0000000060301013 -> 000000006bf05013
Info 391: 'refRoot/cpu', 0x0000000080000050(trap_handler): Machine 34021273 csrrw   x4,mscratch,x4
Info 391: 'refRoot/cpu',   MEMX 0x80000050 0x80000050 2 1273
Info 391: 'refRoot/cpu',   MEMX 0x80000052 0x80000052 2 3402
Info 391: 'refRoot/cpu',   x4 0000000000000000 -> 0000000080002400
Info 391: 'refRoot/cpu',   mscratch 0000000080002400 -> 0000000000000000

```

eroom1966, 2024-07-15:
oh, hang on, I think you may have made updates to the testbench I do not have.
you have changed the variant to RV64GCK

I need to go check, but doesnt this instruction exist as part of the crypto extension ?

davidharrishmc, 2024-07-15:
Yes, I changed the variant to RV64GCK.

grevi was proposed as part of the crypto extension but never ratified.  
I’m concerned that it was incorporated into ImperasDV during a draft spec,
and not removed when the final spec was ratified.

David

davidharrishmc, 2024-07-15:
See, for example, Imperas/ImperasLib/source/riscv.ovpworld.org/processor/riscv/1.0/model/riscvDecode.c

eroom1966, 2024-07-16:
Hi David, having investigated this further - it was an issue when enabling the B and K subsets.
grevi was accidentally being brought in.
I have a fix, and will release shortly, all tests pass running iterelf

davidharrishmc, 2024-08-09:
Lee, how do I get this grevi fix?

eroom1966, 2024-08-12:
Hi David,

Apologies – this slipped off my plate.
I need to get Duncan to cut a release for you – he is on vacation this week, so we should get this early next week

Thx
Lee

eroom1966, 2024-08-20:
Duncan is setting off an eng build today, so you should get a notification by tomorrow

Thx
Lee

davidharrishmc, 2024-09-12:
Verified fixed by August 2024 ImperasDV release

closed 2024-09-12 by davidharrishmc