openhwfoundation/cvw #885: Question on Reset exception

eroom1966, 2024-07-17:
Are the actions associated with a reset documented anywhere

I am simulating against the reference a reset, and we see differences in the following
```
Info (IDV_RPT) SCENARIO 2:
Info (IDV_RPT)  +  Net reset => 1 { when:1262 }
Info (IDV_RPT)     +  Net reset => 0 { when:1274 }
Info (IDV_RPT)        +  Retire { pc:0x80000000(rvtest_entry_point+0), inst:'80000eb7 lui     x29,0x80000' }
Info (IDV_RPT)           ! dut.mtvec        - 0x0000000000000000 BASE:0x0 MODE:0(Direct)
Info (IDV_RPT)             ref.mtvec        - 0x0000000080000080 BASE:0x80000080 MODE:0(Direct) (not updated)
Info (IDV_RPT)           ! dut.mscratch     - 0x0000000000000000
Info (IDV_RPT)             ref.mscratch     - 0xffffffff8b4db30c (not updated)
Info (IDV_RPT)           ! dut.mepc         - 0x0000000000000000
Info (IDV_RPT)             ref.mepc         - 0x00000000800001a6 (not updated)
Info (IDV_RPT)           ! dut.mtval        - 0x0000000000000000
Info (IDV_RPT)             ref.mtval        - 0x00000000d3b7aa0b (not updated)
Info (IDV_RPT)
```
There is no definition in the ISA for the reset of these registers, so they need to be handled explicitly in the reference to match the intention of the RTL.
By observation I see 
- mtvec
- mscratch
- mepc
- mtval
I am assuming I am only seeing these as they are non-zero differences - their may well be others

davidharrishmc, 2024-08-09:
Unfortunately there is no helpful Wally documentation.  The code is the best place to look.

Unless there are obscure cases not coming to mind immediately, all of the Wally CSRs take reset to begin in a deterministic value to make simulation easier to manage.
See, for example, src/privileged/csrm, which has resettable flops (flopenr) on mtvec, scratch, mepc, and mtval.

davidharrishmc, 2024-08-15:
@eroom1966 can you close this if you are satisfied?

eroom1966, 2024-08-21:
I think we can close this for now, its difficult to know definitively what values should be used, but lets assume it is ok for now

closed 2024-08-21 by eroom1966