openhwfoundation/cvw #886: Floating point setting of FS

eroom1966, 2024-07-18:
in the Floating point specification it says the following 
```
If an instruction explicitly or implicitly writes a floating-point register or the fcsr but does not
alter its contents, and FS=Initial or FS=Clean, it is implementation-defined whether FS transitions to Dirty.
```
Can I ask how you have decided to implement this in CVW ?

I am seeing a case where the fflags.NX=1, but mstatus.FS=Initial, if we execute an instruction which generates an inexact result, what should mstatus.FS be set to ?
fcsr is being written, but its contents are not altered

davidharrishmc, 2024-07-18:
In csrsr.sv, writing to the FFLAGS with a csrw or writing to a floating-point register sets FS to dirty, without checking whether the write actually changes any bits in the CSR or FPR.

      end else if (FRegWriteM | WriteFRMM | WriteFFLAGSM) STATUS_FS_INT <= 2'b11; 

So in your question, if you execute a floating point instruction that generates an inexact result and writes to a floating-point destination register, mstatus.FS should be set to 11 (dirty).

eroom1966, 2024-07-24:
Hi David,
I am unable to reproduce this due to an earlier failure regarding Retired instruction reporting during a RESET. 
I will close for now, and reopen once I can try to reproduce

eroom1966, 2024-07-26:
Hi @davidharrishmc @ross144 

re-opening this testcase - I think I have something which shows the issue - are you able to run this on the ImperasTG variant you created today ?
[ImperasTG.tg.4.zip](https://github.com/user-attachments/files/16391912/ImperasTG.tg.4.zip)

I ran this 
wsim ImperasTG generic --tb testbench --lockstep --sim vcs --elf ImperasTG.tg.4.elf

davidharrishmc, 2024-08-09:
The root cause was that FS was set to dirty by FRegWriteM.  The fcvt.wu.s instruction writes an integer register and updates the floating-point flags, so it didn't assert FRegWriteM and didn't set FS dirty.  This is a real bug.

Unfortunately, making this agree with Imperas is a bit tricky because the spec allows FS to be set dirty conservatively, even if none of the floating-point status actually changes.  I think the easiest approach is to make Wally match ImperasDV behavior.  Lee, is there a precise definition of when ImperasDV sets FS dirty?  Is it all instructions that write the FP registers, plus all converts to integers?  Or only instructions that actually cause a bit in the FP register file or FFLAGS register to change?  Or something else?

This is also tricky to test because we need to set FS back to clean after each fp instruction to detect instructions that should have made it dirty but silently didn't.

eroom1966, 2024-08-12:
Hi David
Thanks for the update.

>> because the spec allows FS to be set dirty conservatively, even if none of the floating-point status actually changes

Could you point me at the section of the spec you are referring to here ?

Thx
Lee

eroom1966, 2024-08-12:
> priv-isa-asciidoc_20240411.pdf section 3.1.6.6 Extension Context Status in mstatus Register
>
> Implementations may choose to track the dirtiness of the floating-point register state imprecisely by reporting the state to be dirty even when it has not been modified. On some implementations, some instructions that do not mutate the floating-point state may cause the state to transition from Initial or Clean to Dirty. On other implementations, dirtiness might not be tracked at all, in which case the valid FS states are Off and Dirty, and an attempt to set FS to Initial or Clean causes it to be set to Dirty.
>
> This definition of FS does not disallow setting FS to Dirty as a result of errant speculation. Some platforms may choose to disallow speculatively writing FS to close a potential side channel.

Hi David,

In our installation, do you have a copy of the PDF file. OVP_RISCV_Model_Custom_Extension_Guide ?
Specifically, there is a section on Floating point (5.10 Floating Point)

Does this cover your intended implementation ?
If not please describe your intended implementation, and we can look to add behavior for the FS field if it appears to be a legal option

Field
Type
Description
mstatus_fs_mode
riscvFSMode
This field specifies how the implementation-dependent mstatus.FS field is updated by the model. The possible behaviors are specified by the riscvFSMode enumeration (see section 5.10.2 below):

typedef enum riscvFSModeE {

    RVFS_WRITE_NZ,

    RVFS_WRITE_ANY,

    RVFS_EXECUTE_NOT_S,

    RVFS_EXECUTE_ANY,

    RVFS_ALWAYS_DIRTY,

    RVFS_FORCE_DIRTY

} riscvFSMode;

   mstatus.FS Update Modes
The RISC-V Privileged Specification allows the behavior of the mstatus.FS field to be implementation defined. The model allows the behavior of this CSR field to be configured using the mstatus_fs_mode configuration option as follows:

  1.  RVFS_WRITE_NZ: mstatus.FS will be set to dirty (3) state whenever an FPR is written or a floating point operation signals an exception. Floating point operations that do not write an FPR or cause an exception will not set mstatus.FS. As an example, floating point comparison operations with Normal operands will not set mstatus.FS because these do not signal an exception.
  2.  RVFS_WRITE_ANY: mstatus.FS will be set whenever an FPR is written or a floating point operation could potentially signal an exception. As an example, floating point comparison operations with Normal operands will set mstatus.FS, because those operations could signal an exception with some inputs (e.g. SNaN operands).
  3.  RVFS_EXECUTE_NOT_S: mstatus.FS will be set to dirty (3) state when any floating point instruction retires, except for floating point stores, which do not modify mstatus.FS.
  4.  RVFS_EXECUTE_ANY: mstatus.FS will be set to dirty (3) state when any floating point instruction retires.
  5.  RVFS_ALWAYS_DIRTY: mstatus.FS can only hold either off (0) or dirty (3) states.
  6.  RVFS_FORCE_DIRTY: mstatus.FS is forced to dirty (3) state.

Thx
Lee

eroom1966, 2024-08-15:
> This is great. Hope PR 912 fixes the issue. Our ImperasDV license is down right now, so please test and close this issue if it works for you.

did you see an email from Aimee describing the resolution ?

eroom1966, 2024-08-15:
> This is great. Hope PR 912 fixes the issue. Our ImperasDV license is down right now, so please test and close this issue if it works for you.

Fixed Thanks!

closed 2024-08-15 by eroom1966

davidharrishmc, 2024-08-23:
Finally got Imperas licenses back up and got to the bottom of this one.  The program is making a misaligned access to a non-existent memory location.  Wally thows misaligned and ref throws access fault.

Because there are no caches and Wally only supports misaligned accesses to cachable memory, ZICCLSM_SUPPORTED = 0.  This means that the system will unconditionally raise misaligned exceptions.  According to Table 15 and the note below, the priority of misaligned is higher than access faults if misaligned is taken unconditonally.  Therefore, the ImperasTG configuration of Wally prioritizes address-misaligned above access fault, creating the discrepancy.

I believe Wally is correct in this configuration and ImperasDV needs to be configurated to match.

The relative priority of load/store/AMO address-misaligned and page-fault exceptions is implementation-defined to flexibly cater to two design points. Implementations that never support misaligned accesses can unconditionally raise the misaligned-address exception without performing address translation or protection checks. Implementations that support misaligned accesses only to some physical addresses must translate and check the address before determining whether the misaligned access may proceed, in which case raising the page-fault exception or access is more appropriate.

David