openhwfoundation/cvw #893: MTVEC takes invalid value, locks up on exception

eroom1966, 2024-07-25:
@davidharrishmc @ross144 

I have a strange situation, which I will try to create a testcase - but is difficult at the moment due to our testing methodology, I am not convinced at this moment that this is a real bug, and could be a testing methodology error.

The scenario is as follows

MTVEC is written with a value of 0xFFFFFFC1
This indicates a vectored mode, with the address of the trap handler held in location 0xFFFFFFC0
This value violates your MTVEC alignment requirement of 64 bytes, but is written into the CSR nonetheless

Later when an exception occurs, the core appears to lockup.
I would expect to see a read of location 0xFFFFFFC0, in order to get the trap handler location, then a fetch from that location - but I do not see that happening, in fact I see no further memory accesses once the exception has occurred.

Any idea what could be happening here ?

eroom1966, 2024-07-25:
I should add that in fact I do see the address 0xFFFFFFC0 appearing on the AHB bus, but the HSELRam is not asserted

davidharrishmc, 2024-07-25:
FFFFFFC0 has a bottom 8 bits of 11000000.  That looks like a multiple of 64. Why are you concerned that it violates the MTVEC alignment?  

In the rv64gc config file, memory goes from 0x80000000 to 0x87FFFFFF.  If virtual memory is not active, that MTVEC address is unpopulated and you are locking up with an infinite instruction fetch access loop.

localparam logic [63:0] UNCORE_RAM_BASE  = 64'h80000000;
localparam logic [63:0] UNCORE_RAM_RANGE = 64'h07FFFFFF;

If you want to test with this address, change UNCORE_RAM_RANGE to 64’h7FFFFFFF.

David

eroom1966, 2024-07-25:
> FFFFFFC0 has a bottom 8 bits of 11000000.  That looks like a multiple of 64. Why are you concerned that it violates the MTVEC alignment?
My mistake, my hand calculation was incorrect, apologies for that.

> In the rv64gc config file, memory goes from 0x80000000 to 0x87FFFFFF.  If virtual memory is not active, that MTVEC address is unpopulated and you are locking up with an infinite instruction fetch access loop.
This makes sense, and was exactly what I observed, once we have got into this state, I think the only way to get out is to generate a reset, which is what I will add

closed 2024-07-25 by eroom1966