openhwfoundation/cvw #894: FP instruction with reserved runding mode does not trap

eroom1966, 2024-07-25:
I had the following successful retirements from Wally
RM=5 is reserved and should trap not retire

```
RET,0,151,800002d8,"e5318193 addi    x3,x3,-429     ",x3=000000004e8dce53,,,CSRb00(mcycle)=0000000000000296 CSRb02(minstret)=0000000000000097,
...
RET,0,188,8000036c,"d021ddd3 fcvt.s.l f27,x3,rm5    ",,f27=ffffffffXxxxxxxx,,CSRb00(mcycle)=0000000000000338 CSRb02(minstret)=00000000000000bc,
```

davidharrishmc, 2024-07-25:
I believe both behaviors are legal. However, I will change the decoder to strictly detect illegal rounding modes (rm = 101 or 110) or (rm = 111 and frm = 101, 110, or 111).

David

From the unprivileged ascidoc at 20.2
The behavior of floating-point instructions that depend on rounding mode when executed with a reserved rounding mode is reserved, including both static reserved rounding modes (101-110) and dynamic reserved rounding modes (101- 111). Some instructions, including widening conversions, have the rm field but are nevertheless

Reserved

Rounding Mode

The RISC-V Instruction Set Manual Volume I | © RISC-V

mathematically unaffected by the rounding mode; software should set their rm field to RNE (000) but implementations must treat the rm field as usual (in particular, with regard to decoding legal vs. reserved encodings).

The ratified version of the F spec mandated that an illegal-instruction exception was raised when an instruction was executed with a reserved dynamic rounding mode. This has been weakened to reserved, which matches the behavior of static rounding-mode instructions. Raising an illegal-instruction exception is still valid behavior when encountering a reserved encoding, so implementations compatible with the ratified spec are compatible with the weakened spec.

and from 2.2
The behavior upon decoding a reserved instruction is UNSPECIFIED.

Some platforms may require that opcodes reserved for standard use raise an illegal- instruction exception. Other platforms may permit reserved opcode space be used for non- conforming extensions.

davidharrishmc, 2024-07-25:
Fixed with PR 895.  

Wally originally had a lazy decoder, and we are likely to still find other cases where it is not trapping on reserved instructions.  We are also likely to find WARL fields that don't read legal.

closed 2024-07-25 by davidharrishmc