openhwfoundation/cvw #898: Review SPI clock modes

ross144, 2024-08-06:
Jacob implemented SPI on the FPGA and found issues with the SPI phase and polarity controls.  Need to verify these are all correct.

davidharrishmc, 2024-08-07:
@naichewa are you still reading this?  If so, could you check with @JacobPease about the issue and see if you concur and have a fix?

ross144, 2024-08-08:
There is also another issue here.  The SPI module isn't working over 5Mhz on the Arty A7. I believe the underlying issue is how the SPIIn is sampled by PCLK rather than SPICLK.  The FPGA is inserting a large clock latency between PCLK and SPICLK so because SPIIn is timed relative to SPICLK synthesis doesn't align SPIIn to PCLK.  We could probably solve this by just changing the timing constraint, but I don't think this is the right thing to do.

SPI samples SPIIn on either the rising or falling edge of SPICLK depending on the operating mode.  Since we aren't doing this I don't think the SPI module will work in any other clock mode than transmit on the falling edge and sample on the rising.  I believe the correct action is to sample SPIIn using SPICLK.

JacobPease, 2024-08-21:
I think we can now close this as of #922.

ross144, 2024-09-12:
We are still having issues with SPI.  There are still have a few times when SPIOUT is transmitte don the wrong PCLK edge and as I've developing the fpga ImperasDV co-emulation the SPI device is misbehaving in complex ways.

davidharrishmc, 2024-10-26:
The SCLK Divider is also broken.  It doesn't look like the enable logic will divide correctly:
SCLK = PCLK/(2*(SckDiv + 1))

The early clock enable also doesn't work for SckDiv = 0 because DivCounter + 1 never equals 0

assign SCLKenableEarly = ((DivCounter + 12'b1) == SckDiv);

rosethompson, 2024-11-02:
I believe PR #1052 solves this issue, but does not prove that is fixes the problem.

rosethompson, 2024-11-07:
Resolved with the new SPI module.

closed 2024-11-07 by rosethompson