openhwfoundation/cvw #909: CacheSim Mismatches

jordancarlin, 2024-08-11:


rosethompson, 2024-11-11:
There are two problems. 
1. There is a cache bug in the LRU. When the cache is stalled the wrong set is sent to the lru.  There is a simple fix.
2.  The cache simulator does not understand CBO instructions.  I'm working on a solution for the cbo instruction; however, committing out the 4 tests which use cbo instructions and including my above fix the cache simulator matches Wally.

davidharrishmc, 2024-11-12:
Nice!  Shall we just run the integer tests, or otherwise exclude CBO instructions?

jordancarlin, 2024-11-12:
Skipping the CBO instructions should just require skipping the coverage64gc and wally64priv suites I think.

rosethompson, 2024-11-12:
I'd like it to work with the CBO instructions.

rosethompson, 2024-11-12:
I'm glad I kept working on this as there is another more serious bug in the cache. There is a structural hazard between updates to the SRAM array and reading the same set on the next instruction.  Most of the time this is avoided as writes to the SRAM arrays are multicycle with the cachefsm and the hazard is avoided.  However we are exposed to the issue under the following scenario.
1. CBO.ZERO, CBO.CLEAN, CBO.INVL, CBOFLUSH, STORE, or AMO followed by any memory instruction which will read the dirty bits from the same set. 

I believe the solution is very simple we just need to forward the written dirty bit to the next cycle.

rosethompson, 2024-11-13:
Fixed with PR # 1090.

closed 2024-11-13 by jordancarlin