openhwfoundation/cvw #910: instruction generates illegal exception, but should be a HINT

eroom1966, 2024-08-12:
the following compressed instruction
0x0002 (c.slli64 zero, 0x0)

takes an illegal instruction exception, I suspect this is either because rd==0 or shamt==0, the manual says the following

For RV32C and RV64C, the shift amount **must be non-zero; the code points with shamt=0 are HINTs.**
For all base ISAs, the **code points with rd=x0 are HINTs**, except those with shamt[5]=1 in RV32C.

In this instance rd==0 and shamt==0, therefore this instruction should be a HINT

Thx
[ImperasTG.tg.60.zip](https://github.com/user-attachments/files/16588318/ImperasTG.tg.60.zip)

Lee

davidharrishmc, 2024-08-13:
Good catch. Please confirm PR 912 fixes this, and close if good.

eroom1966, 2024-08-15:
Fixed Thanks !

closed 2024-08-15 by eroom1966