openhwfoundation/cvw #911: Testfloat on VCS failing

jordancarlin, 2024-08-12:
Testfloat support is added for VCS in #908, but it is not passing right now. There seem to be a few different distinct issues going on with it. 

- The final vector in a test file is being incorrectly read as all 1s
```
$ wsim --tb testbench_fp --sim vcs f_ieee_rv32gc add
Error in f32_add_rne.tv
TestNum          0 VectorNum      46464 OpCtrl 6
inputs: ffffffff 3f800000 ffffffff
SrcA: xxxxxxxx
 Res: 7fffffff 00
 Expected: ffffffff 1f
```
- Likely a separate issue, there is an off-by-one error for several test cases. 
  - Reproduce by commenting out f32_add_rne.tz and then running
```
$ wsim --tb testbench_fp --sim vcs f_ieee_rv32gc add
Running f32_add_rz.tv vectors 

Error in f32_add_rz.tv
TestNum          0 VectorNum          2 OpCtrl 6
inputs: 9ede38f7 3f800000 3e7f7f7f
SrcA: xxxxxxxx
 Res: 3e7f7f7f 01
 Expected: 3e7f7f7e 01
```

davidharrishmc, 2024-10-10:
TestFloat now runs with the RISC-V model rather than IEEE, so drop _ieee from the configs above:

wsim --tb testbench_fp --sim questa f_rv32gc add --sim vcs 
fails
but passes with --sim questsa
and fails with a lint error in Verilator

davidharrishmc, 2024-10-10:
Defer fixing testbench_fp on tools other than Questa

closed 2024-10-10 by davidharrishmc