openhwfoundation/cvw #967: Make fails on new installations because of GCC 14

jordancarlin, 2024-09-29:
The `wally-toolchain-installation` script pulls the latest version of [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain), which was recently updated to use GCC 14. This fails to recognize `q` as a valid extension in the `-march` string, which causes it to fail when building several of our tests.

djdiaz, 2024-09-29:
Would it make sense to lock the commit the script uses at https://github.com/riscv-collab/riscv-gnu-toolchain/commit/b488ddbd6cdc7d2b951f4bccacc324818356bfc3 until the issue with 14 can be resolved? That's the last commit before the update.

jordancarlin, 2024-09-29:
Yes. That’s what we are planning to do.

djdiaz, 2024-09-29:
I'm currently stuck on this, so let me know when you make the change, or I can make it when I have a chance.

jordancarlin, 2024-09-29:
Running a test installation now and will push the change later today

jordancarlin, 2024-09-30:
Temporarily pinned in #972. This should be investigated further so that we can switch to GCC 14 at some point down the road.

jordancarlin, 2024-09-30:
@djdiaz You might need to delete your existing installation if you already tried since the script will only install newer versions by default, not older versions.

djdiaz, 2024-09-30:
Thank you! Will it be enough to simply rebuild and install the RISC-V toolchain, or are there downstream dependencies that also need to be rebuilt?

jordancarlin, 2024-09-30:
I haven't tried switching to an older gcc version before so not positive, but I think you should be fine to just rebuild the RISC-V toolchain. elf2hex can be finicky and might also need to be rebuilt.

PR #1021 (2024-10-19): Update to GCC 14
- Update install script to use the latest GCC (14.2)
- Modify coverage test makefile to manually call the different gcc components so that the lack of Q support in top level GCC can be avoided

Resolves #967
closed 2024-10-19 by rosethompson