openhwfoundation/cvw #971: Remove axisdc from Linux directory

jordancarlin, 2024-09-29:
@ross144 Now that the SD card uses SPI, I believe the buildroot-packages can be removed and the Linux Makefile can be simplified accordingly

ross144, 2024-09-29:
This is correct. @JacobPease Can you remove?

JacobPease, 2024-09-30:
Yes, I'll work on that.

jordancarlin, 2024-10-07:
Looks like there are also some bits in `testbench.sv` related to this that can be removed.

JacobPease, 2024-10-07:
Possibly. The SDC stuff that's there is meant to test the SDC peripheral against an SD card model. But the model doesn't have an SPI mode like an actual SD card would, so we can't test with it. @rosethompson Should I remove that stuff? I'm about to push the other stuff soon.

rosethompson, 2024-10-08:
I would like to keep the SD card model for now as I have a feeling in the next week or two I'll be modifying it to debug SPI.

jordancarlin, 2024-10-15:
@rosethompson @JacobPease
As a follow up to this, do we still need the ahbsdc submodule? I can’t tell if it’s used anywhere. The only mention of it that I could find was as an include directory in the FPGA wally.tcl script, but it doesn't look like any of the modules from it are actually used despite it being included.

rosethompson, 2024-10-15:
We can remove axisdc. I'll delete now.

closed 2024-10-15 by jordancarlin