openhwfoundation/cvw #979: Remove old scripts/files

jordancarlin, 2024-10-01:
I started removing old scripts and files in PR #978. These are additional files that can likely be removed or need to be updated, but I'm not sure about all of them. 

- `sim/`
    - [ ] `bp-results`
        - Should this be here? Seems like references to a bunch of log files. @rosethompson 
    - [ ] `buildrootBugFinder.py`
        - Not currently working. Update or remove?
    - [ ] `bpred-sim.py`
        - Not currently working. Update or remove?
    - [ ] `xcelium`
        - Not compatible with `wsim`, not tested regularly. Should this be removed?
    - [ ] `FPBuild.txt`
        - [ ] Move relevant parts to `tests/fp/README.md` and then remove
    - [ ] `questa/fpga-wave.do`
        - Is this used anywhere? Seems like it can be removed. @rosethompson 
- `tests/`
    - [ ] `custom`
        - Many of these are not run. Are they all still needed or should they be pruned down? Any that are retained should probably be part of regression.
- `studies/`
    - [ ] `testgen`
        - All of the scripts depend on `imperas-riscv-tests`, which is not in the repo anymore. Should these be removed?
- `fpga/`
     - [ ] `generator/bootmem.txt`
       - Is this still needed? Isn't the current version generated by the zsbl Makefile?
     - [ ] `sim`
       - Are these all still used?
- [ ] `dvtestplan.md` and `docs/testpan.md` are redundant and should be combined
- [ ] `gitflow.txt`
    - This should either be updated to align with the book (https vs ssh) or removed and just rely on the new user setup in the README and textbook

davidharrishmc, 2024-10-01:
Here’s ones I can comment on:

> I started removing old scripts and files in PR #978 <https://github.com/openhwgroup/cvw/pull/978>. These are additional files that can likely be removed or need to be updated, but I'm not sure about [...]
Remove
> 
>  bpred-sim.py
> Not currently working. Update or remove?
I think this is part of the bencharking chapter and we need to get it fixed.  
> 
>  xcelium
> Not compatible with wsim, not tested regularly. Should this be removed?
We don’t have the license at HMC.  It was an effort to set up, and I’d prefer to keep it. Maybe a README that is isn’t presently being used.
> 
>  FPBuild.txt
>  Move relevant parts to tests/fp/README.md and then remove
Good
> questa/fpga-wave.do
> Is this used anywhere? Seems like it can be removed. @rosethompson <https://github.com/rosethompson>
> tests/
>  custom
> Many of these are not run. Are they all still needed or should [...]
Sure

jordancarlin, 2024-10-01:
Removed many of these with #983. The ones not yet checked off are the ones I'm still unsure about.

rosethompson, 2024-10-01:
I need to go through the tests/custom directory.  I used these to help debug Wally in the early stages of development.  Some are still useful when debugging issues with the branch predictor and bringing up the FPGA.  Most of these can be removed or moved to a different directory.  I'll make a new issue to describe the required changes and then you can check this one off this issue.

jordancarlin, 2024-10-01:
Great. So that just leaves the DV plans (@davidharrishmc you're probably best suited to update/combine these) and the gitflow.txt document (currently this describes a different flow from the textbook).

closed 2024-10-10 by jordancarlin