openhwfoundation/cvw #988: $WALLY/bin/wally-tool-chain-install.sh Fails

eroom1966, 2024-10-03:
Running on Ubuntu 22.04
when attempting to run the script $WALLY/bin/wally-tool-chain-install.sh as a user (not root)
This fails with the following error

######################################################################################################################################################################################################################################################################################################################################################################################################################################################################################Installing Buildroot and Creating Linux testvectors
######################################################################################################################################################################################################################################################################################################################################################################################################################################################################################Buildroot successfully download.
make[1]: *** [support/dependencies/dependencies.mk:27: dependencies] Error 1
make: *** [Makefile:51: Image] Error 2
Error: Buildroot installation failed
Error on line 387 with command [ "${PIPESTATUS[0]}" == 0 ]

eroom1966, 2024-10-03:
current hash
commit 428e9d0a6d3c2e97b967db029fa3279a5108e85b (HEAD -> main, origin/main, origin/HEAD)

eroom1966, 2024-10-03:
Update: it appears that error messages do not come the the console/stdout, I looked in the $RISCV/logs directory and saw this message

jordancarlin, 2024-10-03:
Thanks for reporting this. Any line with the words error or warning should be printed to stdout. I’ll take a look at tweaking the filter to make sure this message makes it through and may add a check at the beginning for this specific path issue.

jordancarlin, 2024-10-03:
Should avoided in the future by #988. Let us know if the installation still fails after updating your PATH.

jordancarlin, 2024-10-07:
@eroom1966 Did rerunning the installation with the modified PATH work?

davidharrishmc, 2024-10-26:
@eroom1966 are you satisfied?  Can we close this?

davidharrishmc, 2024-10-28:
Closing this issue.  @eroom1966 please reopen if you need anything else.

closed 2024-10-28 by davidharrishmc

eroom1966, 2024-10-29:
Apologies, yes all resolved after a call with @rosethompson