openhwfoundation/cvw #1002: Pip package conflict

jordancarlin, 2024-10-10:
With recent updates made to riscv-isac (https://github.com/riscv-non-isa/riscv-arch-test/pull/514), there is now a package dependency conflict between riscv-isac and matplotlib. See error message below.

```
INFO: pip is looking at multiple versions of riscv-isac to determine which version is compatible with other requirements. This could take a while.
Collecting matplotlib>=3.9.0 (from -r /home/runner/work/cvw/cvw/bin/requirements.txt (line 4))
  Downloading matplotlib-3.9.1.post1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (11 kB)
  Downloading matplotlib-3.9.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (11 kB)
ERROR: Cannot install -r /home/runner/work/cvw/cvw/bin/requirements.txt (line 4) and -r /home/runner/work/cvw/cvw/bin/requirements.txt (line 6) because these package versions have conflicting dependencies.

The conflict is caused by:
    matplotlib 3.9.0 depends on packaging>=20.0
    riscv-isac 0.18.0 depends on packaging==19.0

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
Error: python packages installation failed
```

jordancarlin, 2024-10-10:
Possible solutions are to see if an older version of matplotlib meets our needs and works with packaging 19.0 or to request riscv-arch-test to change the requirement to be packaging >= 19.0 instead of locking it to exclusively 19.0.

rosethompson, 2024-10-10:
Crud this is a problem. I'm going to help Lee reinstall the toolchain next week.  Do you have a suggested interrum solution?

jordancarlin, 2024-10-10:
Interrum solution would be to remove matplotlib. For most of what we do it isn't necessary. If you remove it from the `bin/requirements.txt` file it will be skipped.

jordancarlin, 2024-10-10:
I'll run a test installation with that removed to make sure it works while we decide what to do longer term.

jordancarlin, 2024-10-10:
Seems that it goes beyond just matplotlib. They updated a large number of the pip dependencies with PR 514 and for some reason switched many of them to be pinned to older versions. There is at a minimum also a conflict with PyYAML. I'll continue investigating, but we'll likely need some changes to be made upstream. Not sure why so many of the packages are strictly pinned now.

jordancarlin, 2024-10-10:
Officially impossible to resolve without changes upstream. riscv-config and riscv-isac have conflicting dependencies and riscof depends on both of them.

PR #1009 (2024-10-14): Fix Installation
Updates pip installation to resolve #1002 when combined with riscv-non-isa/riscv-arch-test#519.

Also adds a `--no-buildroot` flag to the installation script for faster testing.
closed 2024-10-14 by rosethompson