openhwfoundation/cvw #1003: wally-toolchain-install.sh error

amar6905, 2024-10-13:
Hello everyone,

The error I get when running wally-toolchain-install.sh is this:
[image]

Has anyone came across this problem before?

davidharrishmc, 2024-10-13:
What operating system and version are you using?

amar6905, 2024-10-13:
Hi,
I am using WSL 2 and this is my Ubuntu version:
[image]

jordancarlin, 2024-10-13:
We haven't done any testing using WSL. Can you attach the log file at `/opt/riscv/logs/riscv-gnu-toolchain.log`? Also, what does the system you are running it on look like (memory, cores, etc.)? The error message `g++: fatal error: Killed signal terminated program cc1plus` is often a sign of a lack of system resources.

Also, issue #1002 will likely prevent a fresh install from working at the moment. Hoping to get that resolved soon.

amar6905, 2024-10-13:
Hi,

The processor of my machine is the AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx     2.10 GHz
which is 4 cores and 8 threads.
Memory is 8 GB and SSD is 512 GB.
I have also attached the log here
[riscv-gnu-toolchain.log](https://github.com/user-attachments/files/17354096/riscv-gnu-toolchain.log)

jordancarlin, 2024-10-13:
Can you try setting `NUM_THREADS` to 1 near the top of `wally-tool-chain-install.sh`? This will give a cleaner log file and reduce memory pressure. If that still doesn't work and there is no obvious clue in the log file, then I would recommend opening an issue with https://github.com/riscv-collab/riscv-gnu-toolchain since that is the tool that is failing to install.

amar6905, 2024-10-13:
Okay I will try the following, thank you

amar6905, 2024-10-13:
If I am able to build Newlib cross-compiler directly from https://github.com/riscv-collab/riscv-gnu-toolchain. Would that be sufficient for building core-v-wally or are there other crucial tool dependencies?

stineje, 2024-10-13:
+amar6905

Hi!  The key to the repository (and the book) is that everything is included for development and testing.  It includes the gcc compiler you indicated + other items such as Spike, SAIL,  QEMU and other tools.  Let us know if you get any errors and we are happy to help.  

The machine specifications you listed should be sufficient.  We have good success with Ubuntu LTS versions.  Take care.

All my best,

James

amar6905, 2024-10-13:
@stineje 
Hi stineje,
Thank you for your assistance.
So wally-tool-chain-install.sh is failing for me at line 216, which seems to be making the multilib installation (Linux+Newlib)
make -j "${NUM_THREADS}" 2>&1 | logger $STATUS; [ "${PIPESTATUS[0]}" == 0 ]

As a test, I separately cloned the riscv-gnu-toolchain from here: https://github.com/riscv-collab/riscv-gnu-toolchain and I ran "make" for newlib installation which turned out successful here: 
[image]

I also ran "make linux" for linux compiler installation and it was successful too.

So my question would be, what is the best way to bypass line 216 in wally-tool-chain-install.sh? So I could be able to run all the other subsequent tool installations

I am quite new to shell scripting but does line 216 have any parameter differences compared to the way I manually installed the  riscv-gnu-toolchain by "make" and "make linux"?

davidharrishmc, 2024-10-13:
Try deleting the -j “${NUM_THREADS}” and then do “make clean” and then “make”.
Jordan had suggested something along these lines

That will cause Make to run sequentially rather than in parallel.  Your system has only 8 GB of RAM, so perhaps the parallel installation is overtaxing it.

David Harris

jordancarlin, 2024-10-14:
PR #1005 should automatically limit the number of threads used for installation on systems with limited memory. Once that is merged try running the installation script from scratch and see if that works.

amar6905, 2024-10-16:
@davidharrishmc 
Hi your suggestion helped me a lot, thank you. I was able to successfully build all the toolchains in the script except for "buildroot and linux testvectors" part. The error I am getting is: 
[image]

**make -j 2>&1 | logger $STATUS; [ "${PIPESTATUS[0]}" == 0 ]**

when I echo $dir it is set to the main /cvw/ directory.

when I manually enter **line 395: cd "$dir"/../linux**  - the directory does not exist

is line 395 supposed to be **cd "$dir"/../linux** or **cd "$dir"/linux**?

Thanks

jordancarlin, 2024-10-16:
What was printed right above the error message/what is in the mentioned log file? 

Also, were you running the commands manually or running the script itself? The command in the script that is setting `$dir` sets it to the directory that the current command is running from. If you run the `wally-tool-chain-install.sh` script then this will be the bin directory, so `$dir/../linux` is correct. If you copied the commands into the terminal directly while in the `cvw` directory, then `$dir` will be set to `cvw`, which could be the cause of your error.

amar6905, 2024-10-19:
Hello @jordancarlin I was running the script itself. This is the error message I recieve:
[image]
Also, I looked into buildroot.log. And I realized the main issue seems to be this in line 9 of the log:
[image]

jordancarlin, 2024-10-19:
Your log file looks very jumbled and out of order. Can you upload the full log file here? Also, it looks like you don't have the latest version of install script. Can you make sure to do a git pull before trying again?

amar6905, 2024-10-19:
Hello @jordancarlin , I did run "make --jobs" to see the status of my toolchain completion.
I have attached the log file here. 
[buildroot.log](https://github.com/user-attachments/files/17446760/buildroot.log)

jordancarlin, 2024-10-19:
What is the exact command you used to run the installation script? And are you saying it was already up to date or you updated it and it is up to date now? Because based on the previous screenshots that’s definitely not the latest version. If you have a fork of the repo did you make sure to pull from upstream?

amar6905, 2024-10-20:
@jordancarlin Thank you, now buildroot linux has successfully installed. I merged the upstream changes into my fork. So the linux Makefile got updated. I also updated the buildroot section of the toolchain script to match the new upstream updates and now everything is installing successfully. Also, I realized running buildroot for me as user was not successful so I made sure to run the whole toolchain script as root.

jordancarlin, 2024-10-20:
Glad to hear that pulling the upstream changes in worked and everything is successfully installed.

What was the error you got when trying to install buildroot as a user? I have tested it as a user level install without issue in the past. What path were you installing it too?

amar6905, 2024-10-20:
The error when running "make" for buildroot as a user was referring to line 51 of /linux/Makefile.

Currently, after running the full script as root. My installation path is in /opt/riscv

jordancarlin, 2024-10-20:
To clarify, did you run the Linux Makefile manually or did you run wally-tool-chain-install.sh which called the Makefile itself? If you ran the installation script as root the Linux Makefile would need to be run as root as well. If the entire installation script is run as a user then the Linux Makefile should have been fine as a user too.

amar6905, 2024-10-20:
I ran the linux Makefile manually just as a testing purpose when buildroot was failing for me at line 397 of the toolchain script, and when I did it as user it was failing for me. So when I came to run the script again after merging your upstream updates I thought to run as root

jordancarlin, 2024-10-20:
Got it. Probably misconfigured environment variables then. If running it manually you need to make sure to source setup.sh first.

davidharrishmc, 2024-10-26:
@amar6905 if this issue is resolved for you, can you close it?

davidharrishmc, 2024-10-28:
Closing this issue.  @amar6905 please reopen if you need anything else.

closed 2024-10-28 by davidharrishmc