openhwfoundation/cvw #1007: Wally mismatches ImperasDV trying to write middle bits of mcause

davidharrishmc, 2024-10-14:
When trying to write 0x10 into mcause, ImperasDV accepts the write, but Wally only takes the bottom 4 bits and hence writes 0, causing a disagreement.  We need to figure out how to configure ImperasDV to ignore the unused bits.

wsim rv64gc tests/priv/ZicsrM.elf --lockstepverbose

# Info   x5 000000000000000f -> 0000000000000010
# Info 86: 'refRoot/cpu', 0x0000000080005018(main+18): Machine ff5ff06f j       8000500c
# Info   MEMX 0x80005018 0x80005018 2 f06f
# Info   MEMX 0x8000501a 0x8000501a 2 ff5f
# Info 87: 'refRoot/cpu', 0x000000008000500c(main+c): Machine 34229073 csrw    mcause,x5
# Info   MEMX 0x8000500c 0x8000500c 2 9073
# Info   MEMX 0x8000500e 0x8000500e 2 3422
# Info   mcause 000000000000000f -> 0000000000000010 [Interrupt:0 Code:15(Store/AMO page fault)->16]
# Info (IDV) Instruction executed prior to mismatch '0x8000500c(main+c): 34229073 csrw    mcause,x5'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x000000008000500c main+c):
# Error (IDV) Mismatch 0> CSR 342 (mcause)
# Error (IDV)   . dut:0x0000000000000000 Interrupt:0 Code:0(Instruction address misaligned)
# Error (IDV)   . ref:0x0000000000000010 Interrupt:0 Code:16
#

davidharrishmc, 2024-10-14:
Also need to disable seed register (0x15) in ImperasDV such that it throws an illegal instruction instead of being accessed.

Only 5 bits of scause (0x142) should be writable in ImperasDV.

It looks like ImperasDV implements an mnoise register (0x7a9) that is not part of the ratified spec.  I think this should be removed from the default configuration.
https://mjos.fi/doc/20210903-icmc-n32b-saarinen-modern-trng.pdf

mhpmevent3-31 are not implemented in Wally and should trap in ImperasDV
and mseccfg (0x747) and tselect (7a0) and tdata1 (7a1), 2 (7a2), 3 (7a3), info (7a4), control (7a5)
debug csrs dcsr, dpc, dscratch0-1 (7b0-7b3)

Found and fixed a real bug in mcounterinhibit bit 1 not being tied to 0 in Wally
pmpcfg registers need careful handling - revisit

rosethompson, 2024-10-14:
@davidharrishmc Is tests/priv a pushed directory?

davidharrishmc, 2024-10-14:
PR #202 pending

jordancarlin, 2024-10-14:
Merged

eroom1966, 2024-10-15:
Hi @davidharrishmc 
There is an override called ecode_mask (exception code mask)
This can be used to mask out bits of the register

Thx
Lee

davidharrishmc, 2024-10-15:
Mostly resolved with @eroom1966 help.  mseccfg remaining.

davidharrishmc, 2024-10-15:
@eroom1966 the cpu/debug_mode=none causes accesses to the 7b0-7bf set of debug mode registers to trap, as desired.

However, ImperasDV isn't trapping yet on accesses to machine mode debug-related registers in the range 7a0-7a5.  How do I get it to trap for those too?

davidharrishmc, 2024-10-18:
@eroom1966  following up on my last two mismatches:
1) As I interpret the spec, mseccfg should be optional in the latest spec.  Wally traps if it is accessed.  ImperasDV does not.  Do you agree it should be optional and that Wally behavior should be legal?  If so, is there a configuration option to get that behavior, or could you add one?
2) Wally traps on accesses to 7a0-7af machine mode debug-related registers and by default ImperasDV does not.  cpu/debug_mode=none made ImperasDV trap on 7b0-7bf, but not on 7a0-7af.  Is there another switch I should use?

Thanks!

davidharrishmc, 2024-10-26:
Closing this and moving the last two mismatches to a new issue.

closed 2024-10-26 by davidharrishmc