openhwfoundation/cvw #1010: Assembler Version 2.42 synax error

rosethompson, 2024-10-14:
The newest version of the assembler is causing a syntax issues with our tests/coverage tests.

```
make
riscv64-unknown-elf-gcc -g -o amoAccessFault.elf -march=rv64gqc_zfa_zba_zbb_zbc_zbs_zfh_zicboz_zicbop_zicbom_zbkb_zbkx_zknd_zkne_zknh -mabi=lp64 -mcmodel=medany \
    -nostartfiles -T../../examples/link/link.ld amoAccessFault.S
/tmp/ccz1qBGx.s: Assembler messages:
/tmp/ccz1qBGx.s: Error: local label `"1" (instance number 1 of a fb label)' is not defined
make: *** [Makefile:20: amoAccessFault.elf] Error 1
```

rosethompson, 2024-10-14:
Interesting the error goes away if the file extension is changed to lowercase .s.

rosethompson, 2024-10-14:
Scratch that. There are more errors which pop up if we switch to lowercase .s.

stineje, 2024-10-14:
This is really interesting.  Historically, based on when I worked on F90, the uppercase .S indicates that the file contents should be run through the preprocessor, while the lowercase .s indicates that the file contents should be assembled directly.   Just some extra info just in case.  Let us know if you find anything interesting.

davidharrishmc, 2024-10-14:
I don't see a local label 1 in the version in the repository.  Wonder what is triggering this?

But I am using local labels in some jump, branch, and privileged tests, so we definitely need support for them.

rosethompson, 2024-10-14:
The problem is line 137 of WALLY-init-lib.h. The new assembler doesn't like the 1f

    beqz a1, 1f         # yes, a0 has 1 in bit 31

davidharrishmc, 2024-10-14:
Ah, that was a change I made last night.  It only affects RV32, which I hadn’t tested.
I just pushed #1011, which should fix this.

The label was undefined, and only used in RV32.

David

rosethompson, 2024-10-14:
Closing with David's pull request #1011.

closed 2024-10-14 by rosethompson