openhwfoundation/cvw #1025: Mismatch with ImperasDV for fround.s, froundnx.s

coreyqh, 2024-10-22:
When attempting to perform a `fround.s` or `froundnx.s` in in the rv32gc and rv64gc configurations, Wally mismatches with ImperasDV. Upon inspection of the decimal values of the rounded floating-point results, the reference is correct while Wally is significantly off. 

```
$ wsim rv32gc ${WALLY}/addins/cvw-arch-verif/tests/rv32/ZfaF/WALLY-COV-fround.s.elf --lockstepverbose

# Info 32: 'refRoot/cpu', 0x000000008000007c(rvtest_entry_point+7c): Machine 00012087 flw     f1,0(x2)
# Info   MEMX 0x8000007c 0x8000007c 2 2087
# Info   MEMX 0x8000007e 0x8000007e 2 0001
# Info   MEMR 0x80004000 0x80004000 4 c3f0787f
# Info   f1 0000000000000000 -> ffffffffc3f0787f
# Info 33: 'refRoot/cpu', 0x0000000080000080(rvtest_entry_point+80): Machine 4040f0d3 fround.s f1,f1
^C# Info   MEMX 0x80000080 0x80000080 2 f0d3
# Info   MEMX 0x80000082 0x80000082 2 4040
# Info   f1 ffffffffc3f0787f -> ffffffffc3f08000
# Info (IDV) Instruction executed prior to mismatch '0x80000080(rvtest_entry_point+80): 4040f0d3 fround.s f1,f1'
# Error (IDV) FPR register value mismatch (HartId:0, PC:0x80000080 rvtest_entry_point+80):
# Error (IDV) Mismatch 0> FPR f1
# Error (IDV)   . dut:0xffffffffc3808000
# Error (IDV)   . ref:0xffffffffc3f08000
# Error (IDV) testbench.idv_trace2api.state_compare @ 1410: MISMATCH
```
Here, the value in the source register was 0xc3f0787f (decimal -480.941375732421875). It is rounded by the reference to 0xc3f08000 (decimal -481.0) and while it is incorrectly rounded by Wally to 0xc3808000 (decimal -257.0).  

```
$ wsim rv32gc ${WALLY}/addins/cvw-arch-verif/tests/rv32/ZfaF/WALLY-COV-froundnx.s.elf --lockstepverbose

# Info 22: 'refRoot/cpu', 0x0000000080000054(rvtest_entry_point+54): Machine 00012007 flw     f0,0(x2)
# Info   MEMX 0x80000054 0x80000054 2 2007
# Info   MEMX 0x80000056 0x80000056 2 0001
# Info   MEMR 0x80004000 0x80004000 4 c22d71d9
# Info   f0 0000000000000000 -> ffffffffc22d71d9
# Info 23: 'refRoot/cpu', 0x0000000080000058(rvtest_entry_point+58): Machine 40507053 froundnx.s f0,f0
# Info   MEMX 0x80000058 0x80000058 2 7053
# Info   MEMX 0x8000005a 0x8000005a 2 4050
# Info   f0 ffffffffc22d71d9 -> ffffffffc22c0000
# Info   fflags 00000000 -> 00000001 [NV:0 DZ:0 OF:0 UF:0 NX:0->1]
# Info   fcsr 00000000 -> 00000001 [NV:0 DZ:0 OF:0 UF:0 NX:0->1 FRM:0(RNE)]
# Info (IDV) Instruction executed prior to mismatch '0x80000058(rvtest_entry_point+58): 40507053 froundnx.s f0,f0'
# Error (IDV) FPR register value mismatch (HartId:0, PC:0x80000058 rvtest_entry_point+58):
# Error (IDV) Mismatch 0> FPR f0
# Error (IDV)   . dut:0xffffffffc2040000
# Error (IDV)   . ref:0xffffffffc22c0000
# Error (IDV) testbench.idv_trace2api.state_compare @ 1100: MISMATCH
```
Here, the value in the source register was 0xc22d71d9 (decimal -43.361179351806640625). It is rounded by the reference to 0xc22c0000 (decimal -43.0) while it is incorrectly rounded by Wally to 0xc2040000 (decimal -33.0)

.elf files that caused the mismatch can be found here 
[froundTests.tar.gz](https://github.com/user-attachments/files/17472637/froundTests.tar.gz)

davidharrishmc, 2024-10-24:
Seems to be RV64gc.

jordancarlin, 2024-10-30:
@coreyqh can this be closed now?

closed 2024-10-30 by coreyqh