openhwfoundation/cvw #1027: ImperasDV FPGA mismatch divuw

rosethompson, 2024-10-23:
I found the first possible Wally bug running the FPGA in lock step with ImperasDV.  The mismatching instruction is 

divuw a0, a0, a1

with a0 equal to 0x13f3d00 and a1 to 0x1c2000.  Wally says the answer should be 0xad, but imperasDV says 0xb (looks correct).

The instruction is stalled for hundreds of cycles by external stall.  Then for 1 cycle is unstalled, then stalled again for hundreds of cycles.  Then it finishes with 5 unstalled cycles.

davidharrishmc, 2024-10-23:
I added this scenario to the coverage tests.  It produces b and matches ImperasDV.
Perhaps the problem is related to the external stall interacting with the divider stall?

add to ieu.sv
    # Division test (having trouble with Rose FPGA emulation)
    li a0, 0x13f3d00
    li a1,  0x1c2000
    divuw a0, a0, a1

wsim rv64gc $WALLY/tests/coverage/ieu.elf --lockstepverbose

# Info 20: 'refRoot/cpu', 0x0000000080005000(main): Machine 013f4537 lui     x10,0x13f4
# Info   MEMX 0x80005000 0x80005000 2 4537
# Info   MEMX 0x80005002 0x80005002 2 013f
# Info   x10 0000000000000000 -> 00000000013f4000
# Info 21: 'refRoot/cpu', 0x0000000080005004(main+4): Machine d005051b addiw   x10,x10,-768
# Info   MEMX 0x80005004 0x80005004 2 051b
# Info   MEMX 0x80005006 0x80005006 2 d005
# Info   x10 00000000013f4000 -> 00000000013f3d00
# Info 22: 'refRoot/cpu', 0x0000000080005008(main+8): Machine 001c25b7 lui     x11,0x1c2
# Info   MEMX 0x80005008 0x80005008 2 25b7
# Info   MEMX 0x8000500a 0x8000500a 2 001c
# Info   x11 0000000000000000 -> 00000000001c2000
# Info 23: 'refRoot/cpu', 0x000000008000500c(main+c): Machine 02b5553b divuw   x10,x10,x11
# Info   MEMX 0x8000500c 0x8000500c 2 553b
# Info   MEMX 0x8000500e 0x8000500e 2 02b5
# Info   x10 00000000013f3d00 -> 000000000000000b

rosethompson, 2024-10-23:
Yes. It is specific to the ExternalStall. But I believe there is an actual bug here. The specific scenario.  The divide is stalled for some number of cycle. It is unstalled for 1 cycle and the divider proceeds from the IDLE state through BUSY all the way to DONE.  However rather than wait in DONE it  returns to IDLE because the state register is flushed by FlushE.  The same thing happens again on the next unstalled cycle.  Finally we get a third repetition of the divide but this time the pipeline advances one instruction.  I'm guessing the data is clobbering by three divides.

davidharrishmc, 2024-10-23:
If there were no external stall, could this happen?

If not, then the bug seems like an interaction of external stall and the
divide FSM. Does it need more handshaking from done to idle?  Or do you
think there is a better architecture for the divider fsm.

rosethompson, 2024-10-23:
Interestingly I've tried removing this requirement from the hazard unit and it broke the arch64m divw test but imperasDV does not detect the failure. So I'm either unearthing a bunch of bugs or going down some invalid rabbit hole.

rosethompson, 2024-10-23:
Ahh. I remember when the branch predictor miss classifies the divide as a branch taken we still need the divide to finish as it won't be flushed. (Doesn't explain imperasDV)

rosethompson, 2024-10-23:
What's important is the branch misprediction did not assert FlushE in the simulation, but did on the FPGA. The only difference is ExternalStall.

rosethompson, 2024-10-23:
Here is the summarized problem.  When the divsqrtfsm transitions to DONE, it no longer asserts FDivBusyE which causes the hazard unit to assert FlushE.  Under normal circumstances this is ok as the rest of the pipeline is not stalled so the divide instruction moves to the memory stage.  However, if the pipeline is stalled for another reason, the divide remains the Execute stage and the divide is re-executed corrupting the data.

The FPGA rvvi co-emulation aggressively stalls so it makes sense this occurred.  However, we can probably construct another example of this happening with a cache miss.

rosethompson, 2024-10-23:
I was able to create a test which replicates the exact scenario above but uses IFUStallF instead of ExternalStall.  The hazard unit avoids the bug because IFUStallF isgated by not FlushDCause.
assign StallWCause = (IFUStallF & ~FlushDCause)

I'm going to rebuild the FPGA with the ExternalStall gated by the same logic.  I believe it will resolve my issue.

rosethompson, 2024-10-24:
Confirmed this isn't an issue with Wally, but an issue with ExternalStall.  However, there it did uncover a bug with ImperasDV.  I'll create a separate issue for that.

closed 2024-10-24 by rosethompson