openhwfoundation/cvw #1053: Remove deprecated Linux scripts

jordancarlin, 2024-11-02:
As mentioned in https://github.com/openhwgroup/cvw/pull/1019#issuecomment-2423552575, several of the scripts in the `linux/testvector-generation` directory are no longer supported. At a minimum this includes `genTrace.sh`, but I suspect there are others that are no longer used as well.

rosethompson, 2024-11-04:
Looks like we only use fixBinMem and that's just to reverse the byte order of the QEMU memory dump.

jordancarlin, 2024-11-05:
Mostly resolved with #1062. Still the potential to 
> consolidate further by replacing fixBinMem with objcopy and then move genInitMem.sh into the linux directory.

davidharrishmc, 2024-11-16:
Did this one get resolved?

jordancarlin, 2024-11-16:
Yes. Fixed with #1062 and #1081

closed 2024-11-16 by jordancarlin