openhwfoundation/cvw #1060: SPI tests failing

jordancarlin, 2024-11-04:
It seems that with the recent changes to SPI, WALLY-spi-01.S is now failing in regression for both rv32 and rv64 in all simulators. Error message below:

```
Error on test rv32i_m/privilege/src/WALLY-spi-01.S result         120: adr = 800062f0 sim (D$) 000000e2 signature = 00000062
```

```
Error on test rv64i_m/privilege/src/WALLY-spi-01.S result         120: adr = 00000000800064d0 sim (D$) 00000000000000a2 signature = 0000000000000062
```

JacobPease, 2024-11-04:
I can't seem to recreate this error. I just cloned a new repo and rebuilt the tests and all of the regression tests are passing for both Verilator and Questa.

jordancarlin, 2024-11-04:
Hmm. They were failing for both me and @coreyqh on the latest main branch. I’ll investigate.

jordancarlin, 2024-11-04:
Well. Very confusing but passing for me now. Was failing for both me and @coreyqh on multiple different branches yesterday.

closed 2024-11-04 by jordancarlin